arduino/arduino-ide #591: Changes to programmers.txt are not recognized

per1234, 2021-11-03:
## Describe the bug

The first time a board is selected, its programmers are cached by the IDE. After that, changes to [the `programmers.txt` configuration file of the boards platform](https://arduino.github.io/arduino-cli/dev/platform-specification/#programmerstxt) are not reflected in the IDE's **Tools > Programmer** menu.

## To Reproduce

1. Close all Arduino IDE windows.
1. Clear the **Programmer** menu cache by deleting the `C:\Users\<user name>\AppData\Roaming\arduino-ide` (Windows) or `~/.config/arduino-ide` (Linux) folder.
   This is done in order to ensure there are boards available without a cached **Programmer** menu to compare with the board that has a cache.
1. Start the Arduino IDE.
1. Use Boards Manager to install "Arduino AVR Boards" (if it is not already installed).
1. Select **Tools > Board > Arduino AVR Board > Arduino Uno** from the Arduino IDE menus.
   The **Programmer** menu entries have now been cached for the "Arduino Uno" board, but not for the other boards of the "Arduino AVR Boards" platform that have not yet been selected in the IDE.
1. Open the **Tools > Programmer** menu and take note of the programmers listed there:
   [image]
1. Close all Arduino IDE windows
1. Open `C:\Users\<user name>\AppData\Local\Arduino15\packages\arduino\hardware\avr\<platform version>\programmers.txt` in a text editor.
1. Add a new programmer definition to the file by adding the following lines to the file:
   ```
   foo.name=Added Programmer
   foo.program.tool=avrdude
   ```
1. Modify a programmer definition by changing this line:
   ```
   avrisp.name=AVR ISP
   ```
   to this:
   ```
   avrisp.name=Modified Programmer
   ```
1. Remove a programmer definition by deleting these lines:
   ```
   avrispmkii.name=AVRISP mkII
   avrispmkii.communication=usb
   avrispmkii.protocol=stk500v2
   avrispmkii.program.protocol=stk500v2
   avrispmkii.program.tool=avrdude
   avrispmkii.program.extra_params=-Pusb
1. Save the file.
1. Start the Arduino IDE.
1. Select **Tools > Board > Arduino AVR Board > Arduino Uno** from the Arduino IDE menus.
1. Open the **Tools > Programmer** menu.
   Note that the menu contents don't reflect any of the changes that were made to `programmers.txt`:
   ![image](https://user-images.githubusercontent.com/8572152/140034877-76dbc756-271d-4790-bf92-293618948a08.png)
1. Select **Tools > Board > Arduino AVR Board > Arduino Leonardo** from the Arduino IDE menus.
   This is the first time selecting this board so its Programmers menu cache is populated now.
1. Open the **Tools > Programmer** menu.
   Note that the menu contents do reflect the changes that were made to `programmers.txt`:
   ![image](https://user-images.githubusercontent.com/8572152/140034913-0775a461-32b8-4a5f-8b6e-4130f732f33c.png)
1. Close all Arduino IDE windows.
1. Clear the programmer menu cache by deleting the `C:\Users\<user name>\AppData\Roaming\arduino-ide` (Windows) or `~/.config/arduino-ide` (Linux) folder.
1. Start the Arduino IDE.
1. Select **Tools > Board > Arduino AVR Board > Arduino Uno** from the Arduino IDE menus.
1. Open the **Tools > Programmer** menu.
   Note that the menu contents do reflect the changes that were made to `programmers.txt`:
   ![image](https://user-images.githubusercontent.com/8572152/140035306-54fa5e1b-f8bb-4b00-992e-6de9a49645d9.png)

## Expected behavior

**Tools > Programmer** menu always reflects the contents of the `programmers.txt` files at the time the Arduino IDE was started, just as it is in the classic Arduino IDE.

## Desktop

- OS: Windows 10, Ubuntu 20.04
- Version: 2.0.0-beta.12-nightly.20211028
Date: 2021-10-28T03:03:25.353Z
CLI Version: 0.19.1 alpha [718bbbf2]

## Additional context

Originally reported at https://forum.arduino.cc/t/programmers-txt-change-ignored/862718
---
The `programmers.txt` changes are recognized as expected by Arduino CLI:

```
$ arduino-cli version
arduino-cli.exe alpha Version: 0.19.1 Commit: 718bbbf2 Date: 2021-09-29T08:28:11Z

$ arduino-cli board details -b arduino:avr:uno --list-programmers
Id                     Programmer name
parallel               Parallel Programmer
usbGemma               Arduino Gemma
atmel_ice              Atmel-ICE (AVR)
buspirate              BusPirate as ISP
foo                    Added Programmer
usbasp                 USBasp
arduinoisporg          ArduinoISP.org
usbtinyisp             USBtinyISP
jtag3isp               Atmel JTAGICE3 (ISP mode)
arduinoisp             ArduinoISP
stk500                 Atmel STK500 development board
avrisp                 Modified Programmer
arduinoasisp           Arduino as ISP
jtag3                  Atmel JTAGICE3 (JTAG mode)
arduinoasispatmega32u4 Arduino as ISP (ATmega32U4)
```

---

<details><summary>Complete modified `programmers.txt` file:</summary>

```
foo.name=Added Programmer
foo.program.tool=avrdude

avrisp.name=Modified Programmer
avrisp.communication=serial
avrisp.protocol=stk500v1
avrisp.program.protocol=stk500v1
avrisp.program.tool=avrdude
avrisp.program.extra_params=-P{serial.port}

usbtinyisp.name=USBtinyISP
usbtinyisp.protocol=usbtiny
usbtinyisp.program.tool=avrdude
usbtinyisp.program.extra_params=

arduinoisp.name=ArduinoISP
arduinoisp.protocol=arduinoisp
arduinoisp.program.tool=avrdude
arduinoisp.program.extra_params=

arduinoisporg.name=ArduinoISP.org
arduinoisporg.protocol=arduinoisporg
arduinoisporg.program.tool=avrdude
arduinoisporg.program.extra_params=

usbasp.name=USBasp
usbasp.communication=usb
usbasp.protocol=usbasp
usbasp.program.protocol=usbasp
usbasp.program.tool=avrdude
usbasp.program.extra_params=-Pusb

parallel.name=Parallel Programmer
parallel.protocol=dapa
parallel.force=true
# parallel.delay=200
parallel.program.tool=avrdude
parallel.program.extra_params=-F

arduinoasisp.name=Arduino as ISP
arduinoasisp.communication=serial
arduinoasisp.protocol=stk500v1
arduinoasisp.speed=19200
arduinoasisp.program.protocol=stk500v1
arduinoasisp.program.speed=19200
arduinoasisp.program.tool=avrdude
arduinoasisp.program.extra_params=-P{serial.port} -b{program.speed}

arduinoasispatmega32u4.name=Arduino as ISP (ATmega32U4)
arduinoasispatmega32u4.communication=serial
arduinoasispatmega32u4.protocol=arduino
arduinoasispatmega32u4.speed=19200
arduinoasispatmega32u4.program.protocol=arduino
arduinoasispatmega32u4.program.speed=19200
arduinoasispatmega32u4.program.tool=avrdude
arduinoasispatmega32u4.program.extra_params=-P{serial.port} -b{program.speed}

usbGemma.name=Arduino Gemma
usbGemma.protocol=arduinogemma
usbGemma.program.tool=avrdude
usbGemma.program.extra_params=
usbGemma.config.path={runtime.platform.path}/bootloaders/gemma/avrdude.conf

## Notes about Dangerous Prototypes Bus Pirate as ISP
## Bus Pirate V3 need Firmware v5.10 or later
## Bus Pirate V4 need Firmware v6.3-r2151 or later
## Could happen that BP does not have enough current to power an Arduino board 
## through the ICSP connector. In this case disconnect the +Vcc from ICSP connector
## and power Arduino board in the normal way.
buspirate.name=BusPirate as ISP
buspirate.communication=serial
buspirate.protocol=buspirate
buspirate.program.protocol=buspirate
buspirate.program.tool=avrdude
buspirate.program.extra_params=-P{serial.port}

# STK500 firmware version v1 and v2 use different serial protocols.
# Using the 'stk500' protocol tells avrdude to try and autodetect the
# firmware version. If this leads to problems, we might need to add
# stk500v1 and stk500v2 entries to allow explicitly selecting the
# firmware version.
stk500.name=Atmel STK500 development board
stk500.communication=serial
stk500.protocol=stk500
stk500.program.protocol=stk500
stk500.program.tool=avrdude
stk500.program.extra_params=-P{serial.port}

jtag3isp.name=Atmel JTAGICE3 (ISP mode)
jtag3isp.communication=usb
jtag3isp.protocol=jtag3isp
jtag3isp.program.protocol=jtag3isp
jtag3isp.program.tool=avrdude
jtag3isp.program.extra_params=

jtag3.name=Atmel JTAGICE3 (JTAG mode)
jtag3.communication=usb
jtag3.protocol=jtag3
jtag3.program.protocol=jtag3
jtag3.program.tool=avrdude
# Set a bitclock of 0.1us (the fastest supported value). This should
# work regardless of the crystal used, since JTAG doesn't use the MCU
# clock but dictates its own clock.
jtag3.program.extra_params=-B0.1

atmel_ice.name=Atmel-ICE (AVR)
atmel_ice.communication=usb
atmel_ice.protocol=atmelice_isp
atmel_ice.program.protocol=atmelice_isp
atmel_ice.program.tool=avrdude
atmel_ice.program.extra_params=-Pusb
```

</details>

JAndrassy, 2022-07-02:
at least some workaround? where does it cache the content of programmers.txt?

per1234, 2022-07-02:
The workaround is to delete the folder where the cache is:

### Windows

```text
%APPDATA%\arduino-ide\
```

(e.g., `C:\Users\<user name>\AppData\Roaming\arduino-ide\`)

### Linux

```text
~/.config/arduino-ide/
```

### macOS

```text
~/Library/Application Support/arduino-ide/
```

per1234, 2023-03-12:
Hi @interbiznw. For support with the workaround, please post on the Arduino Forum:

https://forum.arduino.cc/c/software/arduino-ide-2-0/93

I'll be happy to provide assistance over there.

dlitz, 2023-06-05:
I seem to be able to get the programmers list to refresh by doing the following:

1. Running `arduino-ide --open-devtools` to open the Chromium Developer Tools
2. Going to *Application* -> *Local Storage* -> *file://*
3. Searching for "programmers" (it should return some keys starting with `theia:.arduinoIDE-configOptions-arduino:`)
4. Deleting the entry/entries for the board(s) I'm using.
5. Quitting and restarting Arduino IDE

bsdmry, 2024-04-14:
Thanks a lot @dlitz! You saved me a lot of time!