arduino/arduino-ide #592: loads other than .ino files on every other start only

joysfera, 2021-11-03:
**Describe the bug**
A project contains one .ino file and a bunch of other files (.cpp, .h). When the IDE is started it loads the project's .ino file only. Then you quit the IDE, start it again and voila - it loads all the files in the project folder. You quit the IDE again, start it for the third time and suddenly just the .ino file is loaded again. This is 100% reproducible.

**Expected behavior**
All project files are loaded every time.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version: 2.0.0 beta.12

per1234, 2021-11-03:
Hi @joysfera. Thanks for taking the time to report this bug.

I see we already have a prior issue report about this at https://github.com/arduino/arduino-ide/issues/541.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

>Why the bug report form does not ask the most important information - the Arduino IDE version?

That's a good point! I always have used the "Version" field under the "Desktop" section of the issue template for that, but I can see how that could be interpreted as being intended for the OS version.

closed 2021-11-03 by per1234

joysfera, 2021-11-03:
I swear I listed through the open issues and tried to find it before creating a new one. I must have overlooked it, sorry.