arduino/arduino-ide #596: Arduino Portenta is not recognized automatically

pitdagosti, 2021-11-05:
Plugging my Portenta Machine Control using a USB cable to my Windows 10 host, the Arduino IDE 2.0.0-beta.12 does not recognize automatically the device.

**Steps to reproduce the behavior**:
On a Windows system:
1. Open 'Arduino IDE 2.0.0-beta.12'
2. Connect the Arduino Portenta Machine Control
3. See error: nothing happens

**Expected behavior**
After some seconds of connecting the Portenta Machine Control, a message should appear saying:

The "Name of the core" core has to be installed for the currently selected "Name of the device" board. Do you want to install it now?

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 21H1 (build SO 19043.1320)

gmacario, 2021-11-05:
Installazione su PC nemo c/o AROL R&D Torino di Arduino IDE 2.0 beta 12

Scaricato file “Arduino IDE 2.0 beta“ per Linux 64-bit (X86-64) da https://www.arduino.cc/en/software#experimental-software

Logged as gpmacario@nemo

per1234, 2021-11-08:
Hi @pitdagosti. Thanks for your feedback.

>After some seconds of connecting the Portenta Machine Control, a message should appear saying:
...

I'll add some additional context for this expectation. I see that this is the behavior of the classic Arduino IDE. With Arduino IDE 2.x this offer to install the missing boards platform is only triggered after the user selects a board of that platform from the dropdown board menu at the top of the Arduino IDE 2.x window.

pitdagosti, 2021-11-08:
Yes, in fact after I selected the board on  the manager I installed it successfully. I just noticed that for some boards the IDE recognizes the drivers automatically, for others it has to be done manually and I didn't know if I was doing something wrong. Thank you for the feedback!

closed 2021-11-08 by pitdagosti

VibinNoodles, 2021-12-25:
I am having issues connecting my portent it connected at first and I downloaded all the necessary things and now it isn't connecting at all

per1234, 2021-12-25:
>I am having issues connecting my portent it connected at first and I downloaded all the necessary things and now it isn't connecting at all

@VibinNoodles has opened a dedicated issue report about this here: https://github.com/arduino/arduino-ide/issues/718
We will confine any further discussion to that issue report unless we discover that it is on topic for this one (which there is no evidence of at the moment).