arduino/arduino-ide #603: No matching member function for call to 'readBytes'

AndreKR, 2021-11-12:
**Describe the bug**
The function `readBytes()` has a wiggly line under it and in the tooltip there is an error:
> No matching member function for call to 'readBytes'

The code compiles without warnings despite the alleged error.

**To Reproduce**
Steps to reproduce the behavior:
1. Open [this example](https://github.com/espressif/arduino-esp32/blob/b94b38c9d11f3c4a85ac3e18a9ec22132351716d/libraries/HTTPClient/examples/StreamHttpClient/StreamHttpClient.ino) from the official ESP32 Arduino core
2. Look in line 75

**Actual behavior**

There is a wiggly line under `stream->readBytes`:
[image]

**Expected behavior**
No wiggly line.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.0.0-beta.12

cmaglie, 2021-12-09:
@AndreKR I cannot reproduce with the latest language server, but I didn't try with the previous release.

Could you try this build? https://github.com/arduino/arduino-ide/actions/runs/1559080827
Which board did you select?

AndreKR, 2021-12-10:
I'm still seeing it with that build.

[image]

I didn't reinstall the boards because 2.0.1 still seems to be the latest available.

The selected board doesn't seem to make a difference, but it's "ESP32 Dev Module".