arduino/arduino-ide #605: No ports discovered

Silver-Fang, 2021-11-14:
**Describe the bug**
IDE could not discover any board

**To Reproduce**
Steps to reproduce the behavior:
1. Open the IDE
2. Install Arduino AVR Boards
3. Found no ports connected
4. Pull out my USB connected board and plug it in again
5. Still found no ports connected

**Expected behavior**
IDE should have found my Arduino Mega 2560 board.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Professional
 - Version: 19043.1348

**Additional context**
Both windows device manager and MATLAB can discover my board. MATLAB scripts can normally connect to the board. It must be the Arduino IDE to blame.

per1234, 2021-11-14:
It looks very similar to https://github.com/arduino/arduino-ide/issues/573, but there is a significant difference:

> 4. Pull out my USB connected board and plug it in again
> 5. Still found no ports connected

When I experience https://github.com/arduino/arduino-ide/issues/573, after unplugging and then replugging the board its port is always recognized by the IDE (though any other ports on the computer remain unrecognized.

Silver-Fang, 2021-11-15:
> It looks very similar to #573, but there is a significant difference:
> 
> > 1. Pull out my USB connected board and plug it in again
> > 2. Still found no ports connected
> 
> When I experience #573, after unplugging and then replugging the board its port is always recognized by the IDE (though any other ports on the computer remain unrecognized.

Because this is beta.12 not 11?😀 The bug grows up along with the version number😀

per1234, 2021-11-30:
Hello @Silver-Fang. The similar bug https://github.com/arduino/arduino-ide/issues/573 was fixed for me by `builtin:serial-discovery@1.3.0`, which is now in use by the nightly build of Arduino IDE 2.x.

Please try the nightly build and let us know whether the bug is fixed for you or not. The download links for the nightly build are here:
https://github.com/arduino/arduino-ide#nightly-builds

Silver-Fang, 2021-12-14:
> Hi @Silver-Fang. Please try again with the latest release of the Arduino IDE and let us know whether or not the bug is fixed for you.

Yes, it's fixed.

closed 2021-12-14 by Silver-Fang

Asderlop, 2023-01-16:
I have a similar bug. After connecting a Wemos Lolin32 board, the port to which the board is connected is not displayed in the Boards Manager or the Tools menu. Restarting the computer and restarting the IDE does not solve the problem. If I connect an original Arduino Uno board, the IDE sees it and shows the port it is connected to.
Windows 11 | Arduino IDE 2.0.3

per1234, 2023-01-17:
Hi @Asderlop. Please post about your problem over on the Arduino Forum:

https://forum.arduino.cc/c/software/arduino-ide-2-0/93

I'll provide support for your problem over there. If we eventually determine it is caused by a problem in Arduino IDE, I will follow up to make sure it is tracked by the developers.