arduino/arduino-ide #606: Please support C++17 by default, or at least give a convenient menu option in the IDE

Silver-Fang, 2021-11-14:
**Is your feature request related to a problem? Please describe.**
My code contains variable templates that C++11 does not support. So each time I set up Arduino IDE for a new PC/user, I'll have to go to "%AppData%\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\platform.txt" and set the "-std=gnu++17" compiler argument. Isn't C++17 fully compatible with C++11 code? Why is the old C++11 still the default?

**Describe the solution you'd like**
I want the C++17 argument to be the default after installation. 

**Describe alternatives you've considered**
Or at least there should be a menu option in the IDE. It's ugly to dive into such deep directories and find a .txt configuration file to do edition.

**Additional context**

pzygielo, 2021-11-14:
It seems your issue is for [arduino/Arduino](https://github.com/arduino/Arduino) (while this repo is for 2.0). I guess this one could be transferred there by maintainers.

Silver-Fang, 2021-11-14:
> It seems your issue is for [arduino/Arduino](https://github.com/arduino/Arduino) (while this repo is for 2.0). I guess this one could be transferred there by maintainers.

No, this is for 2.0. In 2.0 you still have to manually edit platform.txt to support C++17.

per1234, 2021-11-14:
Thanks for taking the time to submit an issue @Silver-Fang .

I see we already have a prior issue report about this at https://github.com/arduino/arduino-ide/issues/430.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-11-14 by per1234

pzygielo, 2021-11-14:
So it's rather https://github.com/arduino/ArduinoCore-avr then.
Just for reference - previously updated in https://github.com/arduino/ArduinoCore-avr/commit/c84ff4d1778592009724f2078c7f6453a31b1a3f.