arduino/arduino-ide #608: The user is not guided to the error line when an error is encountered after verifying/uploading the sketch

91volt, 2021-11-15:
**Describe the bug**
If an error is encountered when trying to compile or upload, in code editor the user is not guided  to the line of code where the first encountered error is located. This is annoying especially when dealing with long code or multiple .ino tab, because the user is not sure on where the error is located, making it difficult to spot and troubleshoot it.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'File'
2. Go to 'Examples > Strings'
3. Open 'StringReplace' sketch
4. Go to line 46
5. Delete the semicolon ";" at the end of line
6. Scroll on top of code
7. Click Verify (or Upload) button on top bar of IDE GUI

**Expected behavior**
When an error is encountered trying to compile or upload In code editor I expect to be brought with the cursor to the first error found in the workspace. 

**Screenshots**
Actual behaviour - IDE 2 beta12 
[image]

Expected (Java IDE legacy behaviour)
[image]

**Desktop (please complete the following information):**
 - OS: All
 - Version: 2.0.0 beta12

**Additional context**
This behavior was consolidated in Java IDE.

cmaglie, 2021-12-09:
The Java IDE parses the output of the compiler to get the line number:
https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/Compiler.java#L499-L621

The most valuable parts are:

1) the filter to remove the build path from the file paths:
https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/Compiler.java#L509-L513
this will transform the ugly:
`/var/folders/3v/64328736489326472384762843762347/T/arduino_sketch/Blink.ino:15`
to a better:
`Blink.ino:15`

2) the regexp to catch the errors and column/lines:

https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/Compiler.java#L137
https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/Compiler.java#L509-L513
https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/Compiler.java#L580-L595

kittaakos, 2022-06-10:
Upstream issue: https://github.com/eclipse-theia/theia/issues/11284.

commit 3a17550a (2022-06-17): Reveal the error location after on failed verify.

Closes #608
Closes #229
commit 842a078b (2022-06-17): Reveal the error location after on failed verify.

Closes #608
Closes #229
commit 63f1d851 (2022-06-21): Reveal the error location after on failed verify.

Closes #608
Closes #229
commit d6f4096c (2022-06-21): Reveal the error location after on failed verify.

Closes #608
Closes #229
closed 2022-06-21 by kittaakos