arduino/arduino-ide #613: Serial connection doesn't work when multiple IDE windows are open

AlbyIanna, 2021-11-17:
When you open Serial Monitor or Serial Plotter in multiple IDE windows, they all use the same port, even though you have a different serial port selected from the Board Selector or Tools > Port menu in each window.

per1234, 2021-11-17:
Hi @AlbyIanna. Can you confirm that, in regards to Serial Monitor, this is a regression that will only be introduced once https://github.com/arduino/arduino-ide/pull/597 is merged (i.e., it does not occur when using any version of the IDE from the `main` branch at this time)?

AlbyIanna, 2021-11-17:
@per1234 Yes, I confirm it's a bug introduced in the branch `serial-plotter`. We decided to consider this issue an edge case, so we're not planning to solve this before the release. That absolutely does not mean that it's not an important issue, it's just that we have more urgent stuff to do at the moment.

per1234, 2021-11-18:
Even though this issue is believed to be a new regression, just to create a linkage to related information, I'll add that something similar was mentioned incidentally in another issue:
https://github.com/arduino/arduino-ide/issues/584#issue-1040438820

> If a second IDE is open on another port, the one that looses the connection reconnects to the second IDE's port.

per1234, 2022-01-22:
Something that might not be clear from this report is that the Serial Monitors show the merged output from the ports selected in each window. There is a demonstration of that here: https://github.com/arduino/arduino-ide/pull/661#issuecomment-986494324

mklemarczyk, 2022-03-17:
I have same problem, it takes input only from one device and display it in all windows. Even the different boards are connected.
[image]

I uploaded the program to scan Bluetooth only for 33 BLE. The program on IoT should print only line "starting Bluetooth® Low Energy module failed!" or "Movement detected".

wittend, 2022-05-29:
This problem is bedeviling me as well.  On Linux Mint (Ubuntu) I'm trying to sort out two sketches operating as client/server on /dev/ttyACMx (ESP32)  communicating with /dev/USBx (Arduino Atmega4808).