arduino/arduino-ide #617: Cannot edit keybindings

strickce, 2021-11-17:
**Describe the bug**
When I try to edit a keymapping, the input window does not allow me to change the contents.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ArduinoIDE/Advanced/Keyboard Shortcuts
2. Click on any item
3. The edit window is shown
4. Try to type into the field (cannot)
5. Clear the input area, then try to type (still cannot)
6. 
**Expected behavior**
The input area should be enabled for text input.

**Screenshots**
Not needed.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: [12.0.1]  Monterey

**Additional context**
None.

per1234, 2022-06-11:
Hi @strickce. Thanks for your report. Please comment here to say whether the problem stills occurs if you try again using the latest nightly build of the Arduino IDE.

The links to the downloads for the nightly build are listed here:

https://github.com/arduino/arduino-ide#nightly-builds

per1234, 2022-07-10:
Closing due to lack of response.

I am unable to reproduce the issue, using the latest build of the Arduino IDE (2.0.0-rc8-snapshot-7d96153), on both Windows 10 and Ubuntu 20.04 (I don't have access to a macOS machine). So I will have to conclude that the issue has been fixed since the time of the report.

@strickce if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

strickce, 2022-10-11:
The problem is almost fixed.  Thank you!  The keys now show as mac keys, and I am able to edit them.  But when I click the edit pen for a key, the edit window that pops up is populated with the windows representation of the key.  for instance, ADD Cursor Above says that the current value is meta+alt+up instead of the mac command+option+up.

per1234, 2022-10-11:
Thanks for the update.

>the edit window that pops up is populated with the windows representation of the key

I see you created a dedicated issue for that bug at https://github.com/arduino/arduino-ide/issues/618. So I think it is safe to consider this issue resolved, since it was about the inability to edit the keybindings, not about the representations used when editing.

closed 2022-10-11 by per1234