arduino/arduino-ide #619: Preferences Proxy change not worked

butor32, 2021-11-18:
Hi,
**Describe the bug**
If you change proxy setting, and after you want no proxy, the configuration don't change, when you want to load a new library.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Menu -> File -> Preferences...'
2. Click on 'Network' tab
3. Activate manual proxy 
4. Save
5. Restart IDE
6. Go again to 'Menu -> File -> Preferences...'
7. Click on 'Network' tab
8. Set 'No proxy'
9. Save
10. Restart IDE
11. Load a library, the proxy are always define.

**Expected behavior**
Error into ~/.arduinoIDE/arduino-cli.yaml
network:
proxy: http://_/
user_agent_ext: daemon

network:
//proxy: http://_/
user_agent_ext: daemon

And now I can load library ! But into IDE you have a bug

**Desktop (please complete the following information):**
 - OS: [Linux Debian]
 - Version: [Bullseyes]

0ljik, 2022-02-25:
Is the preferences menu removed from 2.0.0-rc3? I only have Exit in File menu. I am stuck here with splash screen taking up the center and the IDE not being able to load anything

per1234, 2022-02-27:
Thanks for the report @butor32!

I was able to reproduce the issue directly via [the Arduino CLI gRPC interface](https://arduino.github.io/arduino-cli/dev/rpc/settings/) (using the excellent [grpcurl](https://github.com/fullstorydev/grpcurl) tool), so I have now transferred this issue to the `arduino/arduino-cli` repository.

## Setup

```text
$ arduino-cli version
arduino-cli.exe  Version: git-snapshot Commit: 2f2cbe23 Date: 2022-02-27T08:01:17Z

$ arduino-cli config set network.proxy foo  # set an arbitrary initial `network.proxy` configuration key value

$ arduino-cli daemon
```

## Reproduce issue via the gRPC interface

```text
$ grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/board.proto \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -proto cc/arduino/cli/commands/v1/common.proto \
  -proto cc/arduino/cli/commands/v1/compile.proto \
  -proto cc/arduino/cli/commands/v1/core.proto \
  -proto cc/arduino/cli/commands/v1/lib.proto \
  -proto cc/arduino/cli/commands/v1/monitor.proto \
  -proto cc/arduino/cli/commands/v1/upload.proto \
  -proto cc/arduino/cli/debug/v1/debug.proto \
  -proto cc/arduino/cli/settings/v1/settings.proto \
  127.0.0.1:50051 \
  cc.arduino.cli.settings.v1.SettingsService.GetAll

{
  "jsonData": "{\"board_manager\":{\"additional_urls\":[]},\"daemon\":{\"port\":\"50051\"},\"directories\":{\"data\":\"C:\\\\Users\\\\per\\\\AppData\\\\Local\\\\Arduino15\",\"downloads\":\"C:\\\\Users\\\\per\\\\AppData\\\\Local\\\\Arduino15\\\\staging\",\"user\":\"C:\\\\Users\\\\per\\\\Documents\\\\Arduino\"},\"library\":{\"enable_unsafe_install\":false},\"logging\":{\"file\":\"\",\"format\":\"text\",\"level\":\"info\"},\"metrics\":{\"addr\":\":9090\",\"enabled\":true},\"network\":{\"proxy\":\"foo\",\"user_agent_ext\":\"daemon\"},\"output\":{\"no_color\":false},\"sketch\":{\"always_export_binaries\":false},\"updater\":{\"enable_notification\":true}}"
}
```

🙂 The `network` key is set to `{"proxy": "foo"}`, as expected.

```text
$ grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/board.proto \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -proto cc/arduino/cli/commands/v1/common.proto \
  -proto cc/arduino/cli/commands/v1/compile.proto \
  -proto cc/arduino/cli/commands/v1/core.proto \
  -proto cc/arduino/cli/commands/v1/lib.proto \
  -proto cc/arduino/cli/commands/v1/monitor.proto \
  -proto cc/arduino/cli/commands/v1/upload.proto \
  -proto cc/arduino/cli/debug/v1/debug.proto \
  -proto cc/arduino/cli/settings/v1/settings.proto \
  -d '{"key": "network", "json_data": "{}"}' \
  127.0.0.1:50051 \
  cc.arduino.cli.settings.v1.SettingsService.SetValue

{

}

$ grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/board.proto \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -proto cc/arduino/cli/commands/v1/common.proto \
  -proto cc/arduino/cli/commands/v1/compile.proto \
[... 80 lines cut ...]

}

$ grpcurl \
  -plaintext \
  -import-path ./rpc \
  -proto cc/arduino/cli/commands/v1/board.proto \
  -proto cc/arduino/cli/commands/v1/commands.proto \
  -proto cc/arduino/cli/commands/v1/common.proto \
  -proto cc/arduino/cli/commands/v1/compile.proto \
  -proto cc/arduino/cli/commands/v1/core.proto \
  -proto cc/arduino/cli/commands/v1/lib.proto \
  -proto cc/arduino/cli/commands/v1/monitor.proto \
  -proto cc/arduino/cli/commands/v1/upload.proto \
  -proto cc/arduino/cli/debug/v1/debug.proto \
  -proto cc/arduino/cli/settings/v1/settings.proto \
  127.0.0.1:50051 \
  cc.arduino.cli.settings.v1.SettingsService.GetAll

{
  "jsonData": "{\"board_manager\":{\"additional_urls\":[]},\"daemon\":{\"port\":\"50051\"},\"directories\":{\"data\":\"C:\\\\Users\\\\per\\\\AppData\\\\Local\\\\Arduino15\",\"downloads\":\"C:\\\\Users\\\\per\\\\AppData\\\\Local\\\\Arduino15\\\\staging\",\"user\":\"C:\\\\Users\\\\per\\\\Documents\\\\Arduino\"},\"library\":{\"enable_unsafe_install\":false},\"logging\":{\"file\":\"\",\"format\":\"text\",\"level\":\"info\"},\"metrics\":{\"addr\":\":9090\",\"enabled\":true},\"network\":{\"proxy\":\"foo\"},\"output\":{\"no_color\":false},\"sketch\":{\"always_export_binaries\":false},\"updater\":{\"enable_notification\":true}}"
}
```

🐛 The `network` key **reverted** to the previous `{"proxy": "foo"}`, rather than being set to `{}` according to the `cc.arduino.cli.settings.v1.SettingsService.SetValue` request.

per1234, 2022-02-27:
I see the transfer of the issue to `arduino/arduino-cli` failed. I've been having that problem all day. I will try the transfer again later.