arduino/arduino-ide #624: Compiler Flags

sstaub, 2021-11-19:
Is there a way to set compiler flags like -Wall ...?
Maybe an option field in preferences would helpfull.
Also please open a discussion forum for simple questions.

per1234, 2021-11-19:
>like -Wall

In the specific case of `-Wall`, yes. This is typically achieved by setting the **File > Preferences > Compiler warnings** setting to "More" or "All".

Here is an example of how it is configured in the "Arduino AVR Boards" platform:
https://github.com/arduino/ArduinoCore-avr/blob/1.8.3/platform.txt#L17-L18
Each boards platform author is free to set arbitrary compiler flags, but this is the most common configuration.

>Also please open a discussion forum for simple questions.

We already have one:
https://forum.arduino.cc/c/software/arduino-ide-2-0-beta/93

plapointe6, 2021-11-25:
Being able to specify project-specific build flags would be a great functionnality. I have seen many persons trying to do this on the internet. The current workaround of redefining board specificly for this is less than ideal.