arduino/arduino-ide #626: Network port connect error never goes away

strickce, 2021-11-19:
**Describe the bug**
I got an error when I tried to do a network (OTA)  download to my ESP8266.  The message  was Unexpected error. Reconnecting NodeMCU 1.0 (ESP-12E Module)[esp8266:esp8266:nodemcuv2] on port 191.168.68.109.   The IDE tries to reconnect once per second and keeps repeating, even after I select a serial port and reload the esp8266 sucessfully.  The error message repeating is the subject of this issue, but I will research the connection error itself and may raise another issue for that.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to connect to a port which causes a connection error, then see that the error message will not go away

**Expected behavior**
there should be a way to dismiss the error message

**Desktop (please complete the following information):**
 - OS: [MACOS]
 - Version: [e.g. 12.0.1]

**Additional context**
The ip address of the 8266 shows correctly in the port selection pulldown. I have the 8266 in STA mode.

strickce, 2021-11-19:
More information:  If I get the error, then select the serial port, the error keeps flashing once per second.  Then if I stop the serial monitor and restart it, the error message goes away.  I think that the only error with the message is that it should go away when the network port is deselected.

per1234, 2022-09-06:
Thanks @strickce. I am able to reproduce the bug when using the version of the IDE from the time of the report: 2.0.0-beta.12. I am not able to reproduce the bug with 2.0.0-rc1 or newer.

I didn't bisect the fix any farther than that, but I am pretty sure it happened at https://github.com/arduino/arduino-ide/pull/597

closed 2022-09-06 by per1234