arduino/arduino-ide #628: Port selection disappeared from the tools menu

strickce, 2021-11-20:
**Describe the bug**
I have been using Beta 12 for several days and have successfully uploaded to my esp8266 several times.  But this morning, the port selection is gone from the tools menu.  When I try to load my sketch, I get an error saying that I must select a port, but of course I cannot, since the menu item is missing.

**To Reproduce**
I have no idea what caused the menu to disappear, so I don't know how you could reproduce it.

**Expected behavior**
The menu item should always be present

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: [MACOS]
 - Version: [12.0.1]

**Additional context**
Add any other context about the problem here.

strickce, 2021-11-20:
I just tried to select another board and port, and got an error that there were no ports found.  I unplugged the usb cable and reinserted it and the port showed up there and the menu item returned.  So I'm not sure what the problem really is.  First, I think the IDE should have known that the port was there.  But if not, I think the menu item should be there and it should report that there are no ports found.

ubidefeo, 2021-11-20:
hi @strickce 
on which version of the IDE are you?
Beta 12 or Nightly?
We have done some refactoring this past week on that specific item, so if it's the Nightly it's something we need to address as a regression

per1234, 2021-11-20:
Thanks for taking the time to submit an issue.

I see we already have a prior issue report about this at https://github.com/arduino/arduino-ide/issues/573.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-11-20 by per1234

strickce, 2021-11-21:
I am using beta 12