arduino/arduino-ide #629: Beta 12 crashes on Linux and produces endless loop

AlfredoCubitos, 2021-11-21:
**Describe the bug**
calling arduino-ide produces the following error:

I replaced my home dir with $HOME

Starting backend process. PID: 9301
[9284:1121/121607.182528:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.

arduino-ide_2.0.0> Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file://$HOME/.theia'
root ERROR (node:9301) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
root INFO Theia app listening on http://localhost:35643.
root WARN The local plugin referenced by local-dir:$HOME/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:$HOME/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:$HOME/.arduinoProIDE/plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.1-beta.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO Deploying backend plugin "cortex-debug@0.3.10" from "$HOME/Devel/Arduino/arduino-ide_2.0.0/resources/app/plugins/cortex-debug/extension/dist/extension"
root INFO Deploying backend plugin "vscode-arduino-tools@0.0.1-beta.1" from "$HOME/Devel/Arduino/arduino-ide_2.0.0/resources/app/plugins/vscode-arduino-tools/extension/dist/bundle"
root INFO Deploying backend plugin "cpp@1.52.1" from "$HOME/Devel/Arduino/arduino-ide_2.0.0/resources/app/plugins/vscode-builtin-cpp/extension"
root WARN Backend ConfigServiceImpl.onStart is slow, took: 53.1 ms
daemon INFO Starting daemon from $HOME/Devel/Arduino/arduino-ide_2.0.0/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
root INFO Deploying backend plugin "json@1.46.1" from "$HOME/Devel/Arduino/arduino-ide_2.0.0/resources/app/plugins/vscode-builtin-json/extension"
root INFO Deploying backend plugin "json-language-features@1.46.1" from "$HOME/Devel/Arduino/arduino-ide_2.0.0/resources/app/plugins/vscode-builtin-json-language-features/extension/client/dist/jsonMain"
root INFO Deploy plugins list took: 57.2 ms
root ERROR Could not start contribution
root ERROR Uncaught Exception: 
root ERROR Error [ERR_IPC_CHANNEL_CLOSED]: Channel closed
    at process.target.send (internal/child_process.js:678:16)
    at $HOME/Devel/Arduino/arduino-ide_2.0.0/resources/app/src-gen/backend/main.js:26:17
daemon INFO Using config file: $HOME/.arduinoIDE/arduino-cli.yaml
daemon INFO arduino-cli version 0.19.1
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon INFO Daemon is now listening on 127.0.0.1:50051...
daemon INFO Daemon is running.
daemon ERROR Failed to listen on TCP port: 50051. Address already in use.
daemon INFO Failed to start the daemon.
daemon ERROR Error: Failed to listen on TCP port. Address already in use.
    at Function.parse $HOME/Devel/Arduino/arduino-ide_2.0.0/resources/app/node_modules/arduino-ide-extension/lib/node/arduino-daemon-impl.js:239:24)
    at Socket.<anonymous> ($HOME/Devel/Arduino/arduino-ide_2.0.0/resources/app/node_modules/arduino-ide-extension/lib/node/arduino-daemon-impl.js:142:43)
    at Socket.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at Socket.Readable.push (_stream_readable.js:224:10)
    at Pipe.onStreamRead (internal/stream_base_commons.js:181:23)

**In this state it produces an endless loop, which produces:**

daemon INFO Restarting daemon in 5 seconds...
daemon INFO Failed to listen on TCP port: 50051. Address already in use.
daemon INFO Daemon exited with exit code: 5.
daemon INFO Restarting daemon in 4 seconds...
daemon INFO Restarting daemon in 3 seconds...
daemon INFO Restarting daemon in 2 seconds...
daemon INFO Restarting daemon in 1 seconds...
daemon INFO Restarting daemon now...

**Desktop**
 - OS: Linux
 - Version: openSuSe Tumbleweed with KDE Plasma

per1234, 2021-11-21:
Hi @AlfredoCubitos. Thanks for taking the time to submit an issue.

I see we already have a prior issue report about this at https://github.com/arduino/arduino-ide/issues/201.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

---

I'll provide you with instructions to configure the IDE to use a free port:

1. Open this file in a text editor:
   ```
   ~/.arduinoIDE/arduino-cli.yaml
   ```
1. Change the port number in the `daemon.port` key to some other port that is not already in use (e.g., `"50052"`).
1. Save the file.

Now try starting the Arduino IDE again as before. It should now start up and work as expected.

kittaakos, 2021-11-22:
> It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

Per, I would have one remark on this. According to the logs, there was an error before the `Address already in use.` popped up:

> [9284:1121/121607.182528:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.

It is just a guess, but the Electron app crashed due to [this](https://bugs.launchpad.net/ubuntu/+source/chromium-browser/+bug/1877173) bug (the user's openSuSe OS seems to match the bug), it failed to close the backend and the client stayed connected to the CLI, then starting the IDE next time failed with the port is already in use error. As I said, it is just a guess, but it might need another check before keeping this issue closed.

Proposed solution: https://www.reddit.com/r/openSUSE/comments/ptqlfu/psa_vscode_and_some_nonpatched_electron/
Possible related Arduino IDE issue: https://github.com/arduino/arduino-ide/issues/142

per1234, 2021-11-22:
Thanks so much Akos!

g3grau, 2021-12-21:
I had to look 5x until I realized that I had the same issue. Starting ./arduino-ide results in a first error about the GPU process, followed by a new prompt, followed by a bunch of errors including "Buffer() is deprecated". No GUI appears but 6 processes remain running in the background. The latest version I checked was arduino-ide_nightly-20211220_Linux_64bit.zip

I remembered a workaround for another Electron app ("heavy ion" seems to match better?).
It also works to start the Arduino-IDE:
`./arduino-ide --disable-gpu-sandbox`

A quick test (compiling and uploading a project to an ESP32) worked fine on my Ryzen laptop (dual graphics with AMD and NVidia, is this causing the issue?). I hope that it will also run on an older and less powerful system (version 1.8.15 consumes incredible 7+ minutes at <<10% CPU load for a tiny program). The debug button looks interesting, I'm curious :)