arduino/arduino-ide #632: Only 1 upload then fails - DUE - strange fix at bottom

WRDRJ, 2021-11-24:
Love some of the new features and changes, keep up the great work. Looking forward to some of the cool features.
**Describe the bug**
IDE 2.0 Beta 12
compiles and uploads one time
I make 1 minor delay change to sketch
compiles
won't upload the second time gives error
No device found on COM11
Upload error: Error: 13 INTERNAL: Failed uploading: uploading error: exit status 1
It still shows the correct info for board and COM in both the IDE and Device manager
No changes were made to any wiring or the device

Works fine in IDE 1.8.16, make change and upload, as many times in a row as I need
Hardware:
DUE-CH340 clone of an actual Arduino DUE
DUE programming port
CH340 USB interface, driver wxh.cn, 11/4/2011, Ver 3.3.2011.11
I have 10 devices with this CH340 USB and they all work fine

**Expected behavior**
It should allow as many uploads in a row.
Weird fix:
After it fails the upload, closing Beta 12 and opening the same sketch without changes in IDE 1.8.16, performing the upload, then closing IDE 1.8.16, and going back to Beta 12 make changes, and it now uploads 1 time. So in order to get it to upload the second time from Beta 12, the board has to be uploaded by IDE 1.8.16. 

**Screenshots**
first upload - compiles and uploads
[image]
second upload compiles, but won't upload
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Windows] = Win7 build 7601, 64bit, 16 Gb RAM, Intel i7, nvidia graphics

**Additional context**
it appears that by going to IDE 1.8.16 in between each upload it must be resetting something (HW/SW) that allows Beta 12 to then upload. Unplugging and plugging the board back in, or hitting the board reset does not allow Beta 12 to do the upload. Changing ports and going back into Beta 12 then changing the port config in Beta 12 does not fix the problem. 

I might be able to ship one of these boards to someone in the US if it would help with debugging/development

Thanks for the exciting features in Ver 2

per1234, 2021-11-24:
Hi @WRDRJ. In order to gather more information that might help us to investigate your report. I'm going to ask you to post the full output from the failed upload when in verbose mode.

Please do this:

1. Select **File > Preferences** from the Arduino IDE's menus.
1. Check the checkbox next to **Show verbose output during:** [] **upload**.
1. Click the **<kbd>OK</kbd>** button.
1. Attempt an upload, as you did before.
1. After the upload fails, right click on the black "Output" pane at the bottom of the Arduino IDE 2.x window.
1. From the context menu, click **Copy All**.
1. Now [paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the copied error output into a reply here in this thread.

WRDRJ, 2021-11-24:
Using board 'arduino_due_x_dbg' from platform in folder: C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Using core 'arduino' from platform in folder: C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Detecting libraries used...
"C:\\Users\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino-sketch-5DBE616A193A46CA851725A23C867087\\sketch\\BlinkWithoutDelay.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino-sketch-5DBE616A193A46CA851725A23C867087\\sketch\\BlinkWithoutDelay.ino.cpp" -o "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino-sketch-5DBE616A193A46CA851725A23C867087\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\builtin\\tools\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino-sketch-5DBE616A193A46CA851725A23C867087\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino-sketch-5DBE616A193A46CA851725A23C867087\\sketch\\BlinkWithoutDelay.ino.cpp" -o "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino-sketch-5DBE616A193A46CA851725A23C867087\\sketch\\BlinkWithoutDelay.ino.cpp.o"
Compiling libraries...
Compiling core...
Using previously compiled file: C:\Users\xxxx\AppData\Local\Temp\arduino-sketch-5DBE616A193A46CA851725A23C867087\core\variant.cpp.o
Using precompiled core: C:\Users\xxxx\AppData\Local\Temp\arduino-core-cache\core_arduino_sam_arduino_due_x_dbg_cd46eb70786794566f9917fd2045fd7e.a
Linking everything together...
"C:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-gcc" -mcpu=cortex-m3 -mthumb -Os -Wl,--gc-sections "-TC:\\Users\\xxxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x/linker_scripts/gcc/flash.ld" "-Wl,-Map,C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino-sketch-5DBE616A193A46CA851725A23C867087/BlinkWithoutDelay.ino.map" -o "C:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino-sketch-5DBE616A193A46CA851725A23C867087/BlinkWithoutDelay.ino.elf" "-LC:\\Users\\xxxx\\AppData\\Local\\Temp\\arduino-ske [... cut ...]
"C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino/bossac.exe" -i -d --port=COM11 -U false -e -w  -b "C:\Users\xxxx\AppData\Local\Temp\arduino-sketch-5DBE616A193A46CA851725A23C867087/BlinkWithoutDelay.ino.bin" -R
Send auto-baud
Set binary mode
readWord(addr=0)=0
readWord(addr=0xe000ed00)=0
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0
version()=
chipId=00000000
Unsupported processor
No device found on COM11
Upload error: Error: 13 INTERNAL: Failed uploading: uploading error: exit status 1

WRDRJ, 2021-11-24:
using standalone, not the installed version of Beta

WRDRJ, 2021-11-24:
This is the results when it uploads correctly, you can see it has trouble right at the readword(addr=0) line
Compilation complete.
Performing 1200-bps touch reset on serial port COM11
"C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino/bossac.exe" -i -d --port=COM11 -U false -e -w  -b "C:\Users\Richard Jagers\AppData\Local\Temp\arduino-sketch-5DBE616A193A46CA851725A23C867087/BlinkWithoutDelay.ino.bin" -R
Send auto-baud
Set binary mode
readWord(addr=0)=0x20001000
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0x285e0a60
version()=v1.1 Dec 15 2010 19:25:04
chipId=0x285e0a60
Connected at 115200 baud
readWord(addr=0)=0x20001000
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0x285e0a60
Atmel SMART device 0x285e0a60 found
write(addr=0x20001000,size=0x34)
writeWord(addr=0x20001030,value=0x40)
writeWord(addr=0x20001020,value=0x20010000)
writeWord(addr=0x400e0a00,value=0x600)
writeWord(addr=0x400e0c00,value=0x600)
Device       : ATSAM3X8
readWord(addr=0)=0x20001000
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0x285e0a60
Chip ID      : 285e0a60
version()=v1.1 Dec 15 2010 19:25:04
Version      : v1.1 Dec 15 2010 19:25:04
Address      : 524288
Pages        : 2048
Page Size    : 256 bytes
Total Size   : 512KB
Planes       : 2
Lock Regions : 32
Locked       : readWord(addr=0x400e0a08)=0x1

WRDRJ, 2021-11-24:
Compilation complete.
Performing 1200-bps touch reset on serial port COM11
"C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino/bossac.exe" -i -d --port=COM11 -U false -e -w  -b "C:\Users\xxxx\AppData\Local\Temp\arduino-sketch-5DBE616A193A46CA851725A23C867087/BlinkWithoutDelay.ino.bin" -R
Send auto-baud
Set binary mode
**readWord(addr=0)=0x20001000**
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0x285e0a60
version()=v1.1 Dec 15 2010 19:25:04
chipId=0x285e0a60
Connected at 115200 baud
readWord(addr=0)=0x20001000
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0x285e0a60
Atmel SMART device 0x285e0a60 found
write(addr=0x20001000,size=0x34)
writeWord(addr=0x20001030,value=0x40)
writeWord(addr=0x20001020,value=0x20010000)
writeWord(addr=0x400e0a00,value=0x600)
writeWord(addr=0x400e0c00,value=0x600)
Device       : ATSAM3X8
readWord(addr=0)=0x20001000
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0x285e0a60
Chip ID      : 285e0a60
version()=v1.1 Dec 15 2010 19:25:04
Version      : v1.1 Dec 15 2010 19:25:04
Address      : 524288
Pages        : 2048
Page Size    : 256 bytes
Total Size   : 512KB
Planes       : 2
Lock Regions : 32
Locked       : readWord(addr=0x400e0a08)=0x1

WRDRJ, 2021-11-24:
the last posted one works, and you can see it has trouble right after the set binary mode
readWord(addr=0)=0x20001000 is the return that worked, but the one that fails has a value of 0