arduino/arduino-ide #633: Advanced search for libraries

BanksySan, 2021-11-24:
**Is your feature request related to a problem? Please describe.**
I'm trying to find the [Servo Library](https://github.com/arduino-libraries/Servo) referenced in the Arduino documentation.  Searching for _Servo_ doesn't find the library though, it might be because there's a lot of servo libraries.  If I could filter the search results by some more parameters then I could tell if the problem was that the library isn't being found or if I just have to keep scrolling.

**Describe the solution you'd like**
Some method of entering more precise search criteria.  The [`library.properties`](https://github.com/adafruit/Adafruit_SoftServo/blob/master/library.properties) file has several fields that could be searched on.

**Describe alternatives you've considered**
* An _Advances Search_ button that opens a form with more input fields.
* A system similar to GitHub's search field syntax (e.g. `is:open is:pr author:BanksySan sort:comments-desc`).

**Additional context**
[image]

BanksySan, 2021-11-24:
*IDE Version Info*

- Version: 2.0.0-beta.12
- Date: 2021-10-07T07:55:54.325Z
- CLI Version: 0.19.1 alpha [718bbbf2]

per1234, 2021-11-24:
Thanks for taking the time to make this valuable suggestion @BanksySan .

I see we already have a prior request for this at https://github.com/arduino/arduino-cli/issues/1535.

Note this comment:
>Even though the motivation for the capability is based on a use case most common to Arduino IDE, it seems to me that the best place to implement it is in Arduino CLI.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-11-24 by per1234