arduino/arduino-ide #634: Large file crashes IDE when pushing to Remote Sketchbook

ubidefeo, 2021-11-26:
**Describe the bug**
When using the Remote Sketchbook feature to push and sync Sketches across IDE 2.0 and Arduino Cloud, 
If I add a file above 100KB (could be a font or image byte array file), the IDE's window crashes and the IDE stops working.

**To Reproduce**
Steps to reproduce the behavior:

1. Go to Remote Sketchbook
2. Sign in if you are not
3. Pull a Sketch if you haven't
4. Sketch > Show Sketch Folder
5. add any small file (could be a `.h` or your own `.h`/`.cpp` Class
6. Push 
7. It will work
8. Add a larger file (> 100KB) such as a custom font or image converted to Bitmap
9. Push
10. IDE will crash (see screenshot)

**Expected behavior**
Either issue a warning or fix the upload limit

**Screenshots**
https://user-images.githubusercontent.com/1894487/143565989-5a6c6035-1d2d-4023-818a-1364de03e107.mp4

**Desktop (please complete the following information):**
 - OS: Mac OS 
 - Version: 2.0.0 Beta 12 (any)

fstasi, 2021-11-29:
@ubidefeo seems like it crashes when a new file (not on cloud) is pushed. Size seems not related based on @AlbyIanna investigation.

ubidefeo, 2021-12-22:
tested and working

kittaakos, 2023-02-22:
Moved from [here](https://github.com/arduino/arduino-ide/issues/1903#issuecomment-1439951012):

> I have noticed IDE 2.0.3 crashes when I pull such a sketch:
> 
> I downloaded the problematic sketch from Create:
> 
> [sketch_feb14c.zip](https://github.com/arduino/arduino-ide/files/10803775/sketch_feb14c.zip)

https://user-images.githubusercontent.com/1405703/220637154-6d7ac62d-97fe-4d02-9cfe-097826927ff8.mp4

commit 23906d94 (2023-02-23): fix: encoding when reading a cloud sketch

Closes #449
Closes #634
commit 3f5eddb0 (2023-02-23): fix: encoding when reading a cloud sketch

Closes #449
Closes #634
commit eca79227 (2023-02-24): fix: encoding when reading a cloud sketch

Closes #449
Closes #634
commit 77213507 (2023-03-02): fix: encoding when reading a cloud sketch

Closes #449
Closes #634
closed 2023-03-02 by kittaakos