arduino/arduino-ide #636: Arduino IDE Boards Manager ESP32 needs updating

han-fastolfe, 2021-11-26:
The Arduino IDE Boards Manager can only install up to version 1.0.6 of the Espressif ESP32 boards from Spring 2021, but in August 2021 version 2.0.0 was released and in November 2.0.1 was as well.

Issues with updated libraries expecting 2.x are starting to appear, like https://github.com/adafruit/Adafruit_BusIO/issues/69 and more will come.

Arduino IDE needs to be updated for ESP32 boards ASAP.

per1234, 2021-11-26:
Hi @han-fastolfe. I have no problem with installing the latest 2.0.1 release of the ESP32 boards platform via the Arduino IDE Boards Manager.

I suspect the problem is that you either don't have the ESP32 package index URL in your IDE preferences, or else an incorrect package index URL that is not updated by the ESP32 platform developers.

If the former, the Arduino IDE 2.x Boards Manager will only show the currently installed version of the platform. So if you have 1.0.6 installed, it will look like only 1.0.6 is offered.

Please try this:

1. Select **File > Preferences** from the Arduino IDE menus.
1. To the "**Additional Boards Manager URLs**" field, add the following URL:
   ```
   https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
   ```
   If you already have a different URL for `package_esp32_index.json`, remove it.
   If you have other URLs in that field, separate them with commas.
1. Click the <kbd>OK</kbd> button.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the Arduino IDE.
   This restart is necessary due to a bug in Arduino IDE 2.x (`https://github.com/arduino/arduino-ide/issues/637`)

Now check Boards Manager. You should see that all the versions up to 2.0.1 are now available for installation.

Please comment here to let us know whether this solves your problem, or if you can still reproduce it after making sure you have the updated package index URL in your preferences.

han-fastolfe, 2021-11-27:
@per1234 I was using https://dl.espressif.com/dl/package_esp32_index.json which I had gotten from Espressif docs.

per1234, 2021-11-27:
OK, that is the cause of your problem. If you check the contents of that JSON file, you'll see that 1.0.6 is the latest platform version provided by that URL. So the behavior you observed from the Arduino IDE is correct.

You should use the current package index URL, as specified in [the official documentation](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html).

closed 2021-11-27 by per1234