arduino/arduino-ide #637: Package index is not downloaded after adding URL to preferences

per1234, 2021-11-26:
## Describe the bug

In order to[ install a 3rd party boards platform via Boards Manager](https://www.arduino.cc/en/guide/cores), the user must first add its [package index](https://arduino.github.io/arduino-cli/dev/package_index_json-specification/) URL to the Arduino IDE's **File > Preferences > Additional Boards Manager URLs** field.

After doing that, they will open Boards Manager and expect to find the platforms from that package index listed in addition to the default platforms of [the primary official package index](https://downloads.arduino.cc/packages/package_index.json), as it works in the classic IDE. However, those platforms will not be shown because the Arduino IDE never downloaded the new package index after the user added its URL to the preferences. The user must restart the IDE in order to trigger the package index download.

## To Reproduce

1. If present, delete the following:
   - `<directories.data>/package_esp32_index.json`
   - `<directories.data>/packages/esp32/hardware/`
1. Start the Arduino IDE.
1. Select **File > Preferences** from the Arduino IDE menus.
1. Add the following URL:
   ```
   https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
   ```
   (The issue is not specific to this package index. I just picked an arbitrary one for the sake of the demo.)
1. Click the <kbd>OK</kbd> button.
1. Watch `directories.data` for some time.
   Note that the expected `package_esp32_index.json` file never appeared.
1. Open Boards Manager
1. Watch `directories.data` for some time.
   Note that the expected `package_esp32_index.json` file never appeared.
1. Look for the expected "**esp32 by Espressif Systems**" platform in the Boards Manager offerings.
   Note that it is not present.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the Arduino IDE.
1. Watch `directories.data`.
   Note that the `package_esp32_index.json` file now appears.
1. Open Boards Manager
1. Look for the expected "**esp32 by Espressif Systems**" platform in the Boards Manager offerings.
   Note that it is now present.

## Expected behavior

Package index is downloaded after adding a new URL to  **File > Preferences > Additional Boards Manager URLs**.

## Desktop

- OS: Windows 10
- Version: 2.0.0-beta.12-snapshot.a090dfe
Date: 2021-11-25T17:24:00.480Z
CLI Version: 0.20.0 [553c6375]

Azzkikin, 2022-03-15:
Are there other answers to this problem? Following the above steps does not load additional boards.

per1234, 2022-03-15:
Hi @Azzkikin please make a post on the Arduino forum with a detailed description of what you have tried and the problem you encountered.

https://forum.arduino.cc/c/using-arduino/installation-troubleshooting/18

I'll make sure to check your post over there.

kittaakos, 2022-06-24:
Breaking: https://github.com/arduino/arduino-ide/commit/7f8b227c3969c74a4c3b733918c1be0079bcfac2#diff-f463d1c276a9080ec522ead62c6dd596e5bf330b093b7d51b71738c9795d0fdaR66-R98

The super class' `init` method listens to the CLI config change events, but the custom `init` implementation of the Core client-provider does not do it.

The rest of the logic looks good:
 - The frontend sends the new 3rd party URLs to the backend,
 - The CLI daemon gets updated,
 - The new values are correctly written to the CLI config file, 
 - But the config change event is ignored, and the index update does not run.

commit 99723a99 (2022-07-02): Update package index on 3rd party URLs change.

Closes #637
Closes #906
commit 2e4719ad (2022-07-04): Update package index on 3rd party URLs change.

Closes #637
Closes #906
commit fd4625cb (2022-07-04): Update package index on 3rd party URLs change.

Closes #637
Closes #906
commit a2bec126 (2022-07-05): Update package index on 3rd party URLs change.

Closes #637
Closes #906
closed 2022-07-08 by kittaakos