arduino/arduino-ide #640: Extra non-functional "Search" icon

per1234, 2021-11-30:
## Describe the bug

The environment produced by previous IDE versions causes an extra "Search" icon to be shown in the sidebar when the latest version of the IDE is first opened. When clicked on, it opens an empty panel.

## To Reproduce

1. Start Arduino IDE 2.0.0-beta.12
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the current Arduino IDE.

You will now see the extra "Search" icon:

[image]

## Expected behavior

Only a single, functional, search icon no matter which previous IDE version I have used.

## Desktop

- OS: Windows 10
- Version: 2.0.0-beta.12-snapshot.dd76f91
Date: 2021-11-29T14:55:28.762Z
CLI Version: 0.20.1 [abb21449]

## Additional context

If I delete `C:\Users\per\AppData\Roaming\arduino-ide` the problem goes away until after I use an older version of the IDE again.

The issue also occurs if I run any build prior to dd76f9180c82b0aa2f4614f747187dc5e4a2105d in place of 2.0.0-beta.12 in step (1) of the instructions above.

fstasi, 2021-11-30:
@per1234 this issue should be solved in the lastest nightly:
- new workspace are loaded correctly, with the correct icon binded to che correct search widget
- sadly, previously affected workspace need a manual refresh, as the elements are cached by theia in user local storage. If that's the case running the infamous  `Reset Workbench Layout` is needed in order to fix this.
[image]

per1234, 2021-11-30:
>this issue should be solved in the lastest nightly

The issue as described above still occurs with the latest nightly (`2.0.0-beta.12-snapshot.dd76f91` is equivalent).

>If that's the case running the infamous Reset Workbench Layout is needed in order to fix this.

Doing that does indeed fix it. So does restarting the IDE or opening a new window.

But that still leaves the less than ideal first experience after a user updates from 2.0.0-beta.12 to 2.0.0-beta.13. Not a huge problem, but I thought I would report in case there might be an easy fix.