arduino/arduino-ide #641: How should this arduino-ide be installed on linux? Missing install script.

avlap, 2021-11-30:
I've downloaded and extracted it in /opt
There is no install script like in the older IDE.
Running ./arduino-ide, doesn't launch anything and doesn't print error messages.

avlap, 2021-11-30:
That goes fast...
FYI for **Processing**, when I install it with sudo in /opt using the install.sh script, everything went well, except it seems I had to run the following command as normal user to get Desktop icon. Not sure if this also applies to stable / former Arduino IDE. I'll check tonight or tomorrow. 

xdg-desktop-icon install /usr/share/applications/processing-pde.desktop

avlap, 2021-11-30:
Yes, it seems the script doesn't add a desktop shortcut icon when running the ./install.sh script of the stable IDE.

I had to run as normal user (not root):
xdg-desktop-icon install /usr/share/applications/arduino-arduinoide.desktop

I wonder if that shouldn't be /usr/**local**/share/applications though. It's a local install I would think.

avlap, 2021-11-30:
https://unix.stackexchange.com/questions/196636/system-wide-desktop-file-installation-for-unpackaged-software
https://wiki.archlinux.org/title/Desktop_entries

A bit confusing. I tend to think that those tools (xdg-tools, desktop-file-install) are for packaged packages (which you can install via apt etc.).

Maybe the best way for third party package is to copy them to the right place (/opt/arduino-<version>/usr/share/applications/arduino-arduinoide.desktop) (?)

And then make a symbolic link from the Desktop folder to the Arduino *.desktop file in /opt/arduino....

But this has to be confirmed by someone more knowledgeable then I am on this topic.

avlap, 2021-11-30:
A symlink from /usr/local/share/applications/arduino-arduinoide.desktop to /opt/arduino..../arduino-arduinoide.desktop, should add arduino to the menu.

A symlink from ~/Desktop to /opt/arduino..../arduino-arduinoide.desktop should add a shortcut on the Desktop to open Arduino.

Would be a  solution, for a system wide installation probably. Like I said, needs confirmation though. I could be wrong.

avlap, 2021-12-02:
I think I would install in /opt instead of /usr/local for a package like Arduino.
https://unix.stackexchange.com/questions/11544/what-is-the-difference-between-opt-and-usr-local

pabloab, 2022-08-04:
Are you against installing [using Flatpak](https://flathub.org/apps/details/cc.arduino.IDE2)?