arduino/arduino-ide #652: Unable to upload sketch

dronecz, 2021-12-01:
**Describe the bug**
I can not upload any sketch to my board. I can just compile, nothing else. 

**To Reproduce**
see screenshot

**Expected behavior**
I should be able to upload my sketch

**Screenshots**
[image]

**Desktop (please complete the following information):**
Version: 2.0.0-rc1
Date: 2021-11-30T13:13:52.071Z
CLI Version: 0.20.1 [abb21449]

per1234, 2021-12-03:
I finally figured out how to reproduce this:

1. Quit the IDE if it is running.
1. Delete the configuration folder (simulating a fresh install):
   Windows:
   ```
   C:\Users\<user name>\AppData\Roaming\arduino-ide\
   ```
   Linux:
   ```
   ~/.config/arduino-ide/
   ```
1. Select the board and port for your Arduino board manually.
   You can do this via **Tools > Board** and **Tools > Port** or via "**Select other board and port...**". Just DO NOT select the board via the main board selector dropdown menu.
1. Attempt an upload.
   You should experience the behavior shown in the screencast above, where triggering an upload does nothing at all.

Uploads will not work until you have selected the board via the main board selector dropdown menu. After you have done that once, you can then select the board and port manually. There is apparently some configuration state essential for uploading that is only attained by selecting a board from the main board selector dropdown menu

dronecz, 2021-12-03:
Thank @per1234 for your testing, but this is not working for me. 

I tried reproduce your guide, but it does not help, as you can see here: 

[image]

Then I thoughed that this can happen because of ESP32 support files, so I renamed `%userprofile%\AppData\Local\Arduino15` to `%userprofile%\AppData\Local\Arduino15_` so it will not take any settings from Arduino IDE v1 and I also discovered that Arduino IDE v2 is storing settings in `%userprofile%\.arduinoIDE` folder so I deleted it as well, as you can see here, but I have still no luck with fixing this. 

[image]

Also, here is log from last (clean) session of Arduino IDE v2.
[2021-12-03_log.log](https://github.com/arduino/arduino-ide/files/7648014/2021-12-03_log.log)
 
Many thanks for your help.

per1234, 2021-12-03:
>I tried reproduce your guide, but it does not help

I apologize if I was not clear. The instructions I provided are not the fix for the bug. They are the instruction to be used for reproducing the bug. That is essential for the developers to investigate it and for the testers to check the eventual fix.

I have provided instructions for a workaround for the bug here:
https://forum.arduino.cc/t/upload-does-nothing/930864/28

dronecz, 2021-12-03:
> I have provided instructions for a workaround for the bug here:

I can confirm that this "fix" works.

[image]

ray-leiter, 2021-12-09:
I have the problem that I can compile but not upload. It was originally on my UNO and ptillisch showed me how to fix it and it worked. Problem is I now have the same problem with my DOIT ESP32 DEV KIT V1 board. I'm using IDE 2.0 rc1 and my board I think uses the CP2102 USB to Serial Bridge. In the IDE, the Tools->Port always show as COM5(CP210x). I can't get it to show the board name. However, every once in a while I manage somehow to get it to work, but I have not idea how I did it!
BTW, I also got it to work on my Raspberry Pi Pico.

closed 2021-12-10 by fstasi