arduino/arduino-ide #656: Selecting sketch file with double-click results in duplicate tabs

per1234, 2021-12-02:
## Describe the bug

Arduino sketches may consist of multiple files, which are shown in the Arduino IDE as tabs. The "**Search**" and "**Sketchbook**" side bar panels allow the selection of the active tab by clicking on its sketch file.

When this selection is done under the following conditions:

- The tab has not been selected since the sketch was opened
- The selection is done via a double-click rather than a single click

Instead of selecting the tab as expected, a duplicate tab is opened.

## To Reproduce

1. Open a multiple file sketch in the Arduino IDE.
   The following sketch is provided for the demonstration: [MultipleFiles.zip](https://github.com/arduino/arduino-ide/files/7639211/MultipleFiles.zip)
1. Open the "**Search**" side bar panel by clicking the magnifying glass icon.
1. Search for the text `hello`
   There will be a search result in each of the files of the demo sketch.
1. Double click on a search from either of the tabs other than the one that is currently selected in the editor.
   For example, if the "**SomeTab2.ino**" tab is selected, then double-click the search result in the "**MultipleFiles.ino**" or "**SomeTab2.ino**" tabs.

Note that there are now two tabs for the file associated with the search result you clicked:

[image]

## Expected behavior

Clicking a file of a currently open sketch in a sidebar panel always selects its tab.

Each sketch file only ever gets a single tab in the editor.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc1-snapshot.f0d9894
  Date: 2021-11-30T16:25:41.414Z
  CLI Version: 0.20.1 [abb21449]

## Additional context

The equivalent procedure will also produce duplicate tabs when using the "**Sketchbook**" side bar panel.

fstasi, 2022-04-01:
@msujew we decided to disable split tabs at all for now

per1234, 2022-05-09:
Bug is still not fixed so this must remain open.

closed 2022-05-31 by msujew

heliophagus, 2022-12-22:
On IDE 2.0.3, I ALWAYS get multiple tabs whenever I open ANY .INO files. Changes made in one tab are reflected in the other tab, so it isn't a huge issue, but it IS annoying.