arduino/arduino-ide #659: Dependent menus not updated after .zip library install

per1234, 2021-12-02:
## Describe the bug

After installing a library via **Sketch > Include Library > Add .ZIP Library...**", the library does not appear in the "**File > Examples**" and "**Sketch > Include Library**" menus until after the IDE is restarted.

## To Reproduce

1. Obtain an library in suitable `.zip` format which is not already installed on your computer: https://github.com/arduino-libraries/ArduinoMDNS/archive/refs/heads/master.zip
1. Select **Sketch > Include Library > Add .ZIP Library...** from the Arduino IDE menus.
1. Select the library `.zip` (`ArduinoMDNS-master.zip`).
1. Click the <kbd>Open</kbd> button.
1. Select **File > Examples** from the Arduino IDE menus.
   Note that the expected "**ArduinoMDNS**" submenu item is not present.
1. Select **Sketch > Include Library** from the Arduino IDE menus.
   Note that the expected "**ArduinoMDNS**" submenu item is not present.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the Arduino IDE.
1. Select **File > Examples** from the Arduino IDE menus.
   Note that the expected "**ArduinoMDNS**" submenu item is now present.
1. Select **Sketch > Include Library** from the Arduino IDE menus.
   Note that the expected "**ArduinoMDNS**" submenu item is now present.

## Expected behavior

Dependent menus are refreshed after `.zip` library install

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc1-snapshot.f0d9894
  Date: 2021-11-30T16:25:41.414Z
  CLI Version: 0.20.1 [abb21449]

## Additional context

These menus are updated after `.zip` library installation in the classic Arduino IDE.

kittaakos, 2022-09-08:
The new APIs should solve this. See somewhat related problem [here](https://github.com/arduino/arduino-ide/issues/53#issuecomment-1218008965) and the new APIs [here](https://github.com/arduino/arduino-ide/pull/1333/files#diff-f463d1c276a9080ec522ead62c6dd596e5bf330b093b7d51b71738c9795d0fdaR424-R426).

closed 2023-01-11 by kittaakos

roseanne09, 2023-02-17:
Error: 13 INTERNAL: Library install failed: library not valid. the error occurs when i try to add a zip library of esp32-camera-master in the add .ZIP library
[image]

per1234, 2023-02-17:
@roseanne09 that is off topic here. Please post about it on the Arduino Forum. I'll help you to install the library over there:

https://forum.arduino.cc/

commit 4f5b849b (2023-01-09): fix: update Examples and Include Library menu after ZIP install

Closes #659
PR #1807 (2023-01-11): fix: update Examples and Include Library menu after ZIP install
### Motivation

After installing a library from a ZIP, IDE2 must update the `File` > `Examples` and `Sketch` > `Include Library` menus to show the freshly installed library (if the selected board is appropriate).

Steps to verify:
 - You have the AVR core installed, start IDE2 and select the Uno board,
 - Check `File` > `Examples` and `Sketch` > `Include Library` menus, `ArduinoMDNS` is not present.
 - Follow the steps from https://github.com/arduino/arduino-ide/issues/659, they're excellent, and the get the library from the ZIP,
 [... cut]
commit b2bf368d (2023-01-11): fix: update Examples and Include Library menu after ZIP install

Closes #659