arduino/arduino-ide #662: Parts of the UI disappear when window is to small

dronecz, 2021-12-03:
**Describe the bug**
Footbar of the IDE disappear when window height is < 545 px. Search icon (top right corner) disappear when width of the window < 560px 

**To Reproduce**
see screenshot

**Expected behavior**
Footbar should be always visible or windows should have some minimal size, after which you could not make it smaller.  

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Windows 10 19043.1348
- Version: 2.0.0-rc1
- Date: 2021-11-30T13:13:52.071Z
- CLI Version: 0.20.1 [abb21449]

kittaakos, 2021-12-03:
> Search icon (top right corner) disappear

Related issue: https://github.com/arduino/arduino-ide/issues/129

dronecz, 2021-12-03:
Thx @kit, I did not find this issue.

per1234, 2022-07-08:
Closing as resolved by https://github.com/arduino/arduino-ide/pull/1151

closed 2022-07-08 by per1234