arduino/arduino-ide #665: Tabs also show the path

VladimirVecera, 2021-12-03:
I use multiple files in the project and unfortunately after upgrading from beta 11 to RC1 I have a problem with the open file names in the project. In addition to the name, the path is also displayed, so the whole tab is nonsense. 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 64bit
 - Version:  RC1

per1234, 2021-12-04:
Hi @VladimirVecera. Thanks for your report! I also have noticed this change, but I have only observed it while I am experiencing this bug: https://github.com/arduino/arduino-ide/issues/656. The duplicate tabs bug occurred prior to https://github.com/arduino/arduino-ide/pull/610, but before that the duplicated tabs only had the file name.

Please provide a detailed set of instructions we can follow to reproduce the issue.

Does this issue occur for you with any sketch (e.g., **File > Examples > 01.Basics > Blink**), or it is only a specific sketch? If the latter, please attach the sketch in a reply here as a `.zip` file.

VladimirVecera, 2021-12-04:
If I open any example from the library, the name in the tab works fine. Only the name is displayed.

However, if I open any sketch, its path will always be displayed in addition to the file name.

per1234, 2021-12-04:
What is the technique you use to open the sketch @VladimirVecera?

VladimirVecera, 2021-12-04:
This is my video with problem tab:
https://user-images.githubusercontent.com/64955913/144704488-7b5f68d3-cf38-4713-8eac-8fed966699af.mp4

per1234, 2021-12-04:
Thanks. That is useful. I can see the duplicated tabs:

[image]

so you are experiencing https://github.com/arduino/arduino-ide/issues/656. But I'll leave this open because I think we should keep an eye out for valid conditions under which this unhelpful path is shown in the tab.

I have only experienced [this problem](https://github.com/arduino/arduino-ide/issues/656) while opening sketch files from the "**Sketchbook**" sidebar panel. I don't think you will have the problem if you open sketches using **File > Sketchbook** or **File > Open** or **File > Recent**.

VladimirVecera, 2021-12-04:
More detail. After opening the sketchbook, the tab will load fine. But as soon as I click the mouse on the project, the path is added to the tab.

https://user-images.githubusercontent.com/64955913/144705297-d67b7f03-e1b1-4847-8b4d-92aea822b1d1.mp4

per1234, 2021-12-15:
There is another report of the path in the tab text here:

https://forum.arduino.cc/t/ide-2-0-0-rc2-inconsistent-tab-naming/936096

This time it is happening with the last open sketch that opens automatically on startup in the Arduino IDE, and only on every other startup. I haven't been able to reproduce that variant yet.

per1234, 2021-12-21:
Another report here:

https://forum.arduino.cc/t/very-incorrect-display-in-the-tab-line/938102

per1234, 2021-12-26:
One of the several things that can cause this issue is upgrading from Arduino IDE 2.0.0-beta.11

1. Start [Arduino IDE 2.0.0-beta.11](https://github.com/arduino/arduino-ide/releases/tag/2.0.0-beta.11)
1. Quit Arduino IDE 2.0.0-beta.11
1. Start a newer version of the Arduino IDE.
   The last sketch is opened with duplicated tabs.

per1234, 2022-07-19:
The developers have made multiple changes that should fix this issue:

- https://github.com/arduino/arduino-ide/pull/1012
- https://github.com/arduino/arduino-ide/pull/1024
- https://github.com/arduino/arduino-ide/pull/1085

closed 2022-07-19 by per1234