arduino/arduino-ide #666: MacOS M1 ARM support

MattElek, 2021-12-04:
**Is your feature request related to a problem? Please describe.**
Currently, the Arduino IDE is only built for MacOS x86, and not MacOS M1 ARM. It is however built for ARM32 and ARM64 Linux.

**Describe the solution you'd like**
A build of the Arduino IDE for ARM64.

**Describe alternatives you've considered**
Rosetta 2 x86 translation works as expected, but it's not native.

**Additional context**
Pyserial seems to have already ironed out most issues on ARM. Additionally, there are already ARM64 Arduino ports for Linux.

per1234, 2021-12-05:
>It is however built for ARM32 and ARM64 Linux.

There are no official ARM Linux builds of Arduino IDE 2.x. Only [the classic Arduino IDE](https://github.com/arduino/Arduino) is built for these host architectures: https://github.com/arduino/arduino-ide/issues/107

ubidefeo, 2021-12-07:
@MattElek 
I have tried to build on M1 but some dependencies break, this is why we can't release for M1 Native yet.
The IDE performs very well on M1 using Rosetta 2, and we'll keep trying to build for M1 in the future, but dependencies and Electron related stuff is a stopper.
We don't use PySerial at all, because the Serial Monitor services are handled through the CLI behind the scenes

SConaway, 2022-02-07:
Hey there, I'm wondering what the current roadblocks are...is it upstream stuff with Theia? If there's any work that's been done, I can try to continue off of that.

fstasi, 2022-02-07:
Hi @SConaway, the current roadblocks are currently related to the theia version, that relies on node 14 and an outdated version of Electron.

We currently have a PR in the making, that updates both of them and *could* enable the M1 native build.
At the moment I can't tell you yet if that PR is enough or if some additional tweaks are needed.

If you want to give us a feedback/help, I'd checkout [this PR](https://github.com/arduino/arduino-ide/pull/791), then - on an M1 mac - run these commands

```
yarn --ignore-engines
yarn rebuild:browser
yarn rebuild:electron
```
If you can post the output that would be a great starting point for us to assess what is the updated state of the build

PaulStoffregen, 2022-02-07:
The [platform index specification](https://arduino.github.io/arduino-cli/0.20/package_index_json-specification/#tools-definitions) has only 6 architecture names defined.

    ARM Linux 32-bit (arm-linux-gnueabihf),
    ARM Linux 64-bit (aarch64-linux-gnu),
    macOS 64-bit (x86_64-apple-darwin14),
    Windows (i686-mingw32),
    Linux 32-bit (i686-linux-gnu),
    Linux 64-bit (x86_64-linux-gnu)

Even if the IDE were built native for M1, the tools it runs to actually compile & upload code do not seem to have any way to support M1 until this is updated.

per1234, 2022-02-07:
>has only 6 architecture names defined.

That is only the list of host values used in the example package index above that statement:

>In the example above avr-gcc comes with builds for

The full list is described by this algorithm:

https://github.com/arduino/arduino-cli/blob/10107d2407c2d9997310fc2e0f22dfd15d48e9a8/arduino/cores/tools.go#L126-L195

So the `packages[*].tools[*].systems[*].host` value for a macOS ARM native build of a tool would be something like `arm64-apple-darwin11`, but note this part:
https://github.com/arduino/arduino-cli/blob/10107d2407c2d9997310fc2e0f22dfd15d48e9a8/arduino/cores/tools.go#L185-L191
```go
case "darwin,arm64":
	// Compatibility guaranteed through Rosetta emulation
	if regexpMac64.MatchString(f.OS) {
		// Prefer amd64 version if available
		return true, 20
	}
	return regexpMac32.MatchString(f.OS), 10
```

I had aspirations of providing comprehensive documentation for the supported `packages[*].tools[*].systems[*].host` values in the Package Index Specification, but never figured out how to go about it.

SConaway, 2022-02-08:
> Hi @SConaway, the current roadblocks are related to the theia version, that relies on node 14 and an outdated version of Electron.
> 
> Luckily, we have a PR in the making, that updates both of them and _could_ enable the M1 native build. At the moment I can't tell you yet if that PR is enough or if some additional tweaks are needed.
> 
> If you want to give us a feedback/help, I'd checkout [the PR](https://github.com/arduino/arduino-ide/pull/791), then - on an M1 mac - run these commands
> 
> ```
> yarn --ignore-engines
> yarn rebuild:browser
> yarn rebuild:electron
> ```
> 
> If you can post the output that would be a great starting point for us to assess what is the updated state of the build

Hey @fstasi, wasn't sure where to reply with logs. 

Everything went fine until the actual `yarn start` running on node 14: 
```
$ yarn start
yarn run v1.22.17
$ yarn --cwd ./electron-app start
$ theia start --plugins=local-dir:../plugins
dyld[4888]: Library not loaded: @rpath/libffmpeg.dylib
  Referenced from: /Users/steven/arduino-ide/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Electron Framework
  Reason: tried: '/Users/steven/arduino-ide/node_modules/electron/dist/Electron.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libffmpeg.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e')), '/Users/steven/arduino-ide/node_modules/electron/dist/Electron.app/Contents/MacOS/../Frameworks/libffmpeg.dylib' (no such file), '/Users/steven/arduino-ide/node_modules/electron/dist/Electron.app/Contents/MacOS/../Frameworks/libffmpeg.dylib' (no such file), '/usr/local/lib/libffmpeg.dylib' (no such file), '/usr/lib/libffmpeg.dylib' (no such file)
/Users/steven/arduino-ide/node_modules/electron/dist/Electron.app/Contents/MacOS/Electron exited with signal SIGABRT
✨  Done in 0.76s.
```

SConaway, 2022-02-12:
So, I am looking into things and it seems that the issue is the fact that theia wants node v14 to be used. Because of this, `node-gyp` doesn't build for `arm64e` but `x86_64`. using node v16 solves the issue!

n0thing2speak, 2022-02-26:
> 

so can you release your compiled version?

SConaway, 2022-02-27:
> > 
> 
> so can you release your compiled version?

no, I can't release things for arduino...

n0thing2speak, 2022-03-10:
> 

just a download link for your compiled version

fstasi, 2022-03-10:
@SConaway if you found a workaround for node-gyp compiling for x86_64 I'll create an official (signed) release and upload it.

per1234, 2022-09-08:
Closing as duplicate of https://github.com/arduino/arduino-ide/issues/408

The lack of support for building the IDE on (as opposed to distributing builds for) Apple M1/M2 machines is tracked here: https://github.com/arduino/arduino-ide/pull/933

closed 2022-09-08 by per1234