arduino/arduino-ide #668: Log files in project folder

willie68, 2021-12-06:
**Describe the bug**
There are several log files in the project directory, which should not be there.

**To Reproduce**
Simply start the IDE and open a project. Than yoiu will see 4 log files
inols.log, inols-clangd.log, inols-clangd-err.log, inols-err.log

**Expected behavior**
Log files are ok, but not in the project folder. Should be somewhere in the temp folder with the possibility to remove them. (Manually, maybe automatically, if they are to old.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 11

per1234, 2021-12-06:
Hi @willie68. These are the Language Server debug logs. By default, Language Server logging is turned off, so you must have enabled it.

These logs are only needed for development and troubleshooting of the Arduino IDE and Arduino Language Server.

Due to only being needed for development purposes, "**Enable language server logging**" was removed from the **File > Preferences** in Arduino IDE 2.0.0-rc1. It is now controlled solely via the advanced user interface so regular users will not be likely to inadvertently enable them.

Since they will not be used under ordinary circumstances, and since they are sketch-specific, I don't think the current location of the logs is a significant problem.

---

Here are the instructions for disabling/enabling Language Server logging:

## If using Arduino IDE 2.0.0-beta.12 or older

1. Select **File > Preferences** from the Arduino IDE menus.
2. Uncheck the box next to "**☑ Enable language server logging**"
3. Click the <kbd>OK</kbd> button.

## If using Arduino IDE 2.0.0-rc1 or newer

1. Open the command palette (<kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd>)
2. Select "**Settings: Open Preferences**".
3. Under the "**Arduino > Language: Log**", section, uncheck the box next to "**☑ True if the Arduino Language Server should generate log files into the sketch folder. Otherwise, false. It's false by default.**"

closed 2021-12-06 by per1234

willie68, 2021-12-06:
Hmm, ok, thanks for your explanation. 
default should be false, but in my case it was true... never used it before, or is it used in 1.8.15?
Just for documentation: in the new IDE it's called "Preferences: Open Settings (JSON)" in json format 
or "Preferences: Open Preferences"

[image]

per1234, 2021-12-06:
>default should be false, but in my case it was true.

I did quite a bit of testing with fresh installations and never was able to reproduce this. I'll keep an eye out for other reports of that because it is something that should only be enabled on demand when it is necessary to troubleshoot the Arduino Language Server.

If anyone can provide me with a detailed set of instructions I can follow to reproduce the setting defaulting to enabled, that would be very useful.

>is it used in 1.8.15?

No. The classic Arduino IDE does not use a language server at all and it also has a completely separate preferences storage system. So none of your settings from the classic IDE effect Arduino IDE 2.x

>in the new IDE it's called "Preferences: Open Settings (JSON)" in json format
or "Preferences: Open Preferences"

Thanks for the correction! I wrote the instructions with the hope that others with the same problem might find and use them, but they won't be so helpful if by giving incorrect information. I forgot that the name changed in the Arduino IDE 2.0.0-rc1 release. I was going off the naming from 2.0.0-beta.12 and earlier.

willie68, 2021-12-08:
Again, thank for your explanation. I disable the Log files and it seems to work.

per1234, 2021-12-17:
Hi @icehousedoo.

I wanted to ask you about something you mentioned in passing in your other issue:

https://github.com/arduino/arduino-ide/issues/699#issue-1082956369

>Sketch / Show sketch folder shows files as expected inc 4x log files

I am guessing that the four log files you refer to have these file names:

- `inols-clangd-err.log`
- `inols-clangd.log`
- `inols-err.log`
- `inols.log`

These are the log files for the language server used by the Arduino IDE. They are intended for troubleshooting problems.

The issue here is a report of the language server logging being enabled by default. It is intended to be disabled by default, so you should only see these files in your sketch if you have enabled the language server logging in your IDE preferences, following the procedure I described [here](https://github.com/arduino/arduino-ide/issues/668#issuecomment-986569434).

Did you enable the logging in your IDE preferences? If so, then all is well. If not, then I would be interested in investigating this further since we definitely don't want the IDE to produce these files under ordinary circumstances.

icehousedoo, 2021-12-17:
> #699 (comment)

Hi per1234,
Yes language logs was enabled, apologies, just installed RC2 so it must
have picked the setting up from a previous version (also have used VS Code).
Will respond to your other questions in the forum.

Cheers

Isaac

per1234, 2021-12-17:
Thanks so much for your quick response @icehousedoo!

>apologies, just installed RC2 so it must
have picked the setting up from a previous version

No reason for apologies. You are welcome to leave the logging on if it is useful to you (see the information [here](https://github.com/arduino/arduino-ide/issues/668#issuecomment-986569434) for the new method for controlling the setting). I am only trying to keep an eye out for any other reports of the IDE enabling logging by default since that would be a bad experience for the target Arduino user.