arduino/arduino-ide #670: Language server results not updated after changes to installed libraries

per1234, 2021-12-07:
## Describe the bug

The information provided by the Arduino language server is affected by the dependencies used by the sketch.

The lack of an update of the language server results after a change is made to the installed libraries or platforms (i.e., install, update, or uninstall) may cause them to be incorrect.

## To Reproduce

1. Open the following sketch in the Arduino IDE:
   ```cpp
   #include <Arduino_APA102.h>
   Arduino_APA102 leds(1, 2, 3);
   void setup() {}
   void loop() {}
   ```
   If the library is not installed, problems will be indicated as expected.
1. Install the "Arduino_APA102" library via Library Manager.

Note that the problems are still indicated even though the sketch is now valid:

[image]

## Expected behavior

Language server results are updated after making any change to installed libraries via the IDE interface:

- Boards Manager
- Library Manager
- **Sketch > Include Library > Add .ZIP Library...**

## Desktop

- OS: Windows 10
- IDE Version: 2.0.0-rc1-snapshot.be55a0a
Date: 2021-12-03T11:09:57.266Z
CLI Version: 0.20.1 [abb21449]
- [Arduino Language Server](https://github.com/arduino/arduino-language-server) version: 0.5.0-rc6

cmaglie, 2021-12-09:
With version 0.5.0-rc6 of the LS, the error disappears as soon as the user touch (makes a change) in the source code. This means that the LS needs a signal from the IDE that something changed in the global installation, so to fix this issue we need changes in the IDE.

I see only two possible solutions to this problem:

a) the IDE must restart the language server after each install/uninstall operation (this may be forced by running the command "Arduino: restart language server" from the command palette)

b) the IDE must send a "rebuild" command in some way to the language server

I personally prefer the solution "b" because restarting the language server is an expensive operation. The problem is that there is nothing like a "rebuild" command in the LSP (language server protocol). A workaround could be to produce a bougs "textDocument/didSave" event (that do not actually save anything) that in turns triggers a rebuild on the LS.

kittaakos, 2021-12-10:
> I personally prefer the solution "b" because restarting the language server is an expensive operation.

👍 This seems to be the proper way.

> The problem is that there is nothing like a "rebuild" command in the LSP (language server protocol)

FYI, it's possible to extend it, if you decide to do so in the future. Here is the Java example from VS Code to the JDT-LS:
 - VS Code: https://github.com/redhat-developer/vscode-java/blob/c0d043e53057550e3075713f9c2f2d0fdc4902cf/src/standardLanguageClient.ts#L337-L338
 - LS: https://github.com/eclipse/eclipse.jdt.ls/blob/791f30ed63f9960d5ad7dc2e9ba3b9e203cf2b9a/org.eclipse.jdt.ls.core/src/org/eclipse/jdt/ls/core/internal/lsp/JavaProtocolExtensions.java#L68-L69

> "textDocument/didSave"

Or with [`textDocument/didChange`](https://microsoft.github.io/language-server-protocol/specifications/specification-3-17/#textDocument_didChange) with an empty `contentChanges` but with an increased `version` number.

ubidefeo, 2022-09-02:
@kittaakos @cmaglie 
what's the status of this issue?

PR #1808 (2023-01-16): fix: restart LS on lib/core change, client re-init
### Motivation

The running LS must be restarted on library/platform install/uninstall events and core client re-init events (the latter is fired when the sketchbook location changes in IDE2) so that IDE2 can provide the correct Arduino language features to the users.

### Change description

- force restart of the LS on core/lib changes,
- and sketchbook location change

https://user-images.githubusercontent.com/1405703/211280536-5b183aeb-3ca7-4ef8-b02b-3d8f729d17aa.mp4

### Other information

Closes #670

### Reviewer checklist

* [ ] PR add [... cut]
commit 432f3654 (2023-01-16): fix: restart LS on lib/core change, client re-init

Closes #670
closed 2023-01-16 by kittaakos