arduino/arduino-ide #678: Minimap not correctly rendered

sstaub, 2021-12-12:
**Describe the bug**
When changing the size of the minimap, the size is correctly rendered.
After restart the minimal is rendered with size 1, independent from the settings.

**To Reproduce**
Steps to reproduce the behavior:
1. in cmd palette Open Preferences
2. change minimal scale to e.g. 3
3. close the app
4. restart the app

**Expected behavior**
The size of the minimal should follow the preferences on startup.

**Desktop (please complete the following information):**
 - OS: macOS Monterey
 - Version: 2.0 rc1

ubidefeo, 2021-12-13:
hi @sstaub 

Thank you for reporting this, but as of now mini-map is not part of our restricted set of supported features.
When a user decides to enable advanced Theia's features it is unfortunately upon themselves to manage what the feature breaks or changes.

As we move forward we might decide to implement further functionalities and make them part of IDE 2.0 's feature-set, but now is too early to put this in our roadmap, even though it's a small thing.
Hope you'll understand :)

thanks
ubi

sstaub, 2021-12-13:
Hi @ubidefeo 
It make no sense to develop a new IDE when you cut all the features which are common standards on all newer IDEs like VSC, Theia ...