arduino/arduino-ide #683: Make Boards/Library Manager more helpful regarding version numbers

per1234, 2021-12-13:
There are three cases where a [semver](https://semver.org/)-compliant version number conveys especially important information:
- Pre-release (`MAJOR.MINOR.PATCH-PRERELEASE`)
- Beta (`0.MINOR.PATCH`)
- Major version increment (indicates breaking API change)

My proposals:

By default, don't provide update notifications for pre-release versions. This could be a preferences.txt-only option since only advanced users should want to enable pre-release notifications.

When the user does a Boards/Library Manager installation of a pre-release, beta, or major version increment, display a confirmation dialog that explains the implications:

>Warning: The library version 1.2.3-rc1 you have chosen to install indicates a pre-release version, which is intended only for beta testing.

>Warning: The library version 0.1.2 you have chosen to install indicates the library is still undergoing initial development.

>Warning: The library version 2.0.0 you have chosen to install is a major version increment from the previously installed version. This indicates there have been breaking changes to the API.

References:
- https://github.com/bblanchon/ArduinoJson/issues/787
- https://github.com/esp8266/Arduino/issues/5554

KurtE, 2022-11-02:
Thanks @per1234,

Yes this would be very useful for working with the Teensy boards, where we typically do several beta releases as a way to allow those of us who are working on features to have a common code base to continue work with.  But the typical user who only wishes to only work with stable stuff should not, necessarily see when a new beta release comes out, and worse off, have them automatically updated to the beta, when the notification comes up at program startup saying one or or boards are out of date, and you are given a default option to update all.

Different variations of this issue keep coming up on forum threads like, just now:
https://forum.pjrc.com/threads/71370-Teensy4-1-reading-from-Serial-Monitor-of-Arduino-IDE?p=315178&viewfull=1#post315178

And I commented about it up on the Arduino forum thread:
https://forum.arduino.cc/t/invalid-fqbn-not-an-fqbn/1046630/5

PaulStoffregen, 2022-11-03:
FWIW, for Teensy I've started publishing the non-release versions with major number 0, as that seems to be the least-bad way available today for a single platform index.

So today Teensy's package index has:

1.57.1 -- latest stable
1.57.0
1.65.1
0.58.2 -- latest beta

It's less than ideal for people wishing to use the very latest beta, since IDE 2.0.1 prompts them (too aggressively) to upgrade to 1.57.1.