arduino/arduino-ide #686: wrong arch

koendv, 2021-12-14:
**Describe the bug**
When I build arduino-ide-2.0.0-rc2 on raspberry pi/arm64, keytar.node is compiled as Intel 80386, instead of as ARM aarch64.
```
./electron/build/node_modules/keytar/build/Release/obj.target/keytar.node: ELF 64-bit LSB pie executable, ARM aarch64, version 1 (SYSV), dynamically linked, BuildID[sha1]=3a1f5bfabface1ce95cb96c7f2ab16360d58137a, not stripped
./electron/build/node_modules/keytar/build/Release/keytar.node: ELF 32-bit LSB pie executable, Intel 80386, version 1 (SYSV), dynamically linked, BuildID[sha1]=5297fc4f4077850c4730db5527ea7f40de21e038, stripped
```
See attachment [file.txt](https://github.com/arduino/arduino-ide/files/7711618/file.txt)

Manually copying an Arm aarch64 version of keytar.node over the i386 version allows a sucessful build.

**To Reproduce**
On raspberry pi, execute attached build script.
[build-ide.txt](https://github.com/arduino/arduino-ide/files/7711609/build-ide.txt)

**Expected behavior**
That a native compilation produces a native binary.

**Desktop (please complete the following information):**
 - OS: linux
 - Version: 5.10.63-v8+
 ```
$ uname -a
Linux raspberrypi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
```

**Additional context**
I provide builds of [Arduino-ide-2.0.0 for raspberry pi](https://github.com/koendv/arduino-ide-raspberrypi/). No action is needed, I just want to give  a 'heads up' message to others who wish to compile on arm64.

abusr, 2021-12-21:
You copy keytar.node

from electron/build/node_modules/keytar/build/Release/obj.target/keytar.node
over electron/build/node_modules/keytar/build/Release/keytar.node

is it corect?
I'll try a build in the holidays, I hope.

koendv, 2021-12-21:
On Tue, 21 Dec 2021 05:54:23 -0800
Nat ***@***.***> wrote:

> You copy keytar.node
> 
> from
> electron/build/node_modules/keytar/build/Release/obj.target/keytar.node
> over electron/build/node_modules/keytar/build/Release/keytar.node
> 
> is it corect?

Yes. 
The last step in the build process is: yarn --cwd ./electron/packager/ package
During this step, I repeatedly run: find . -name 'keytar.node' -exec file \{\} \;
As soon as the i386 binary shows up, I do the copy.

I vaguely remember encountering a line that said "if the architecture is not X86_64, then it's intel 32-bit", and thinking that was the culprit.

koen

koendv, 2022-03-03:
Works fine now.

closed 2022-03-03 by koendv