arduino/arduino-ide #690: Arduino IDE 2.0-rc2 - cannt start from File Explorere on Linux

NormanDunbar, 2021-12-15:
**Describe the bug**
The `arduino-ide` application has a file type of 'shared object' and as such, cannot be executed from within a file manager, only from the command line in a terminal session.

**To Reproduce**
Steps to reproduce the behavior:
1. Download IDE 2.0 rc2 (or whatever version) for linux 64 bit.
2. Extract the file.
3. Open the extracted folder in a file manager.
4. Double-click on `arduino-ide`
5. "Unknown file type" apears.
6. View the file's details in the file manager, and it is listed as a shared object, not an application or program.

**Expected behavior**
Double click arduino-ide in my file explorer to open the IDE.

**Screenshots**
Not applicable.

**Desktop (please complete the following information):**
 - OS:Linux Mint 64-bit
 - Version: 20.2 "Uma"

**Additional context**
According to https://docs.arduino.cc/software/ide-v2/tutorials/getting-started/ide-v2-downloading-and-installing#linux, the new 2.0 IDE can only be executed from a terminal session. This is a bug, however, I have a possible fix.

If I view the extracted files in the file explorer, I see that the `arduino-ide` file shows up as a shared object. The same is true in a terminal session with the `file` command:

```
$ file arduino-ide
arduino-ide: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=7836c459ba61a264a29878f37b81cd4047746e3a, stripped
```

I've had this problem since updating`gcc` some time back, and discovered that there is an option when linking to make the output an executable file, not a shared object. The option is `-no-pie`.

If you compile with that option, then the new IDE will be able to be executed from the file manager.

Kind regards,
Norm.

NormanDunbar, 2021-12-15:
Hi @per1234 

yes, I'm fine with those suggestions, thanks. I was trying to helpfully point out a solution (assuming gcc or course) that would avoid the need to create desktop files or one line shell scripts to execute the IDE. I see from the first link you gave me that this need for a one liner has been noted.(by @pert on 6th March) but even at RC2, there is no script. 

I'm happy messing about in the bowels of Linux, it's what I do, but other beginners may not be so clued up and if you can't get the "obvious" executable to run from the file manager, then many will possibly give up there and then.

Take care.

Cheers,
Norm.

bperrybap, 2021-12-15:
How is this still an issue?
I get having an issue like this for an alpha release, not really understanding how it could exist in a  beta release.
And how can any development team call this an RC release candidate when, as shipped, there is no way to even start the IDE from the linux GUI file manager?

In my world of s/w development RC candidates mean that the development team believes that the s/w is feature complete, there  are no serious show stopper issues. and this image could be the actual release.
I would call not being able to start the application a serious show stopper issue.

Given there are some really simple solutions to this, why has not been fixed?

bperrybap, 2021-12-15:
For example, we are spending more time writing about the issue and tracking than the amount of time to fix it.
That seems kind of silly.

ubidefeo, 2021-12-30:
@bperrybap 

I'm really sorry this grinds your gears so much.
We're trying to do our best to bring a stable release to our users, but if you're familiar with the world of open-source, when it comes to products with millions of users and hundreds of issues, we are forced to assign priority and criticality in order to let a team of just a handful of developers address them.
I agree with you that RC means it's almost ready for prime time, but if you use the Java IDE you'll still notice things that won't work.
I have just added criticality and priority labels, this will give it a more prominent position in our backlog.
Thank you for participating to the conversation

Ubi

bperrybap, 2021-12-30:
I do apologize that I may have come off as being a bit harsh. My comments were intended to be constructive criticism intended to raise the level of attention to get the issue resolved.
My criticism was that the s/w was being presented as a Release Candidate and it clearly is not at that level yet given a more casual user can't even bring up the application from the OS GUI file manager.
IMO, that type of issue should be a highest priority show-stopper type bug that moves right to the top of the list to be fixed.
Also, I presented a simple solution to this issue back in March. (a 1 line shell script wrapper), but this issue has been allowed to continue to linger for more than 9 months after it was known and there was a simple solution.
It is the combination of all this that bothered me.

Many linux users these days are similar to Windows users and simply don't have the technical skills to solve an issue like this.
They are more casual type users that become quickly overwhelmed if it requires more than just clicking on something from the GUI file manager or Window manager task bar.
As far as myself, it is not much more than an irritation as I have over 40 years of embedded development experience and approximately the same experience with unix and now linux and open source development so fixing an issue like this for myself is trivial and I do fix this issue in my own installations for each of these 2.x releases.
But for many users it can be a show stopper.

ubidefeo, 2021-12-30:
@bperrybap no need to apologise, really
We have very tight schedules and this project is massive.
As you can see I have finally added labels which will bump this up in the queue.

I have been following this silly, easy to fix issue for a while, it's just that sometimes giving priority to everything is not working well.
I know you've been discussing this for months, and I appreciate you not letting it slide.

In the end we're all here to make a better product, you included :)
Our team is comprised of 2 developers on the frontend and 2-3 on the backend, plus Per sees everything and reports any activity, and I jump in and out of GH and Forums as much as I can and triage issues on a daily basis

I'm sorry that the RC is still exhibiting these glitches (I know, this is more than a "glitch"), and I understand how they can be a show-stopper for beginners.
The reason to release an RC was to prompt for more feedback than we could get just through our Beta users, and fortunately we're mostly seeing all the same issues and glitches we already have lined up.

The new year is going to bring more fixes, more RC(s) and soon a Final Release.
Hold tight ✌🏼

bperrybap, 2022-03-05:
here is the arduino executable wrapper script I use on linux mint.
It is more than a simple/single line since the gnome gui file managers like caja do not properly handle the current working directory when you click on an executable. (I've had arguments with them over this) 

```
#!/usr/bin/env bash
#=============================================================================
# bash script to start up Arduino executable
#=============================================================================

#get name & location of this script
progwd=`dirname $0`

#make sure script running where script lives
cd $progwd

# start up Arduino IDE
exec ./arduino-ide
```

per1234, 2022-03-05:
Although the requested functionality is still not available when using the ZIP format Arduino IDE for Linux, it is now provided by the newly added (https://github.com/arduino/arduino-ide/pull/846) [AppImage format](https://appimage.org/) Arduino IDE for Linux.

[The 2.0.0-rc4 and release and onward](https://github.com/arduino/arduino-ide/releases/latest) are [published](https://www.arduino.cc/en/software#future-version-of-the-arduino-ide), as well as the [tester builds](https://github.com/arduino/arduino-ide/actions/workflows/build.yml?query=event%3Apull_request) and [nightly builds](https://github.com/arduino/arduino-ide#nightly-builds) in AppImage format in addition to the previous ZIP format.

bperrybap, 2022-07-25:
RC9 and this is still an issue.
What is the hold up on fixing this?

benlumia007, 2022-09-29:
Is there any updates on this feature

bperrybap, 2022-09-29:
With the the 2.0.0 release on linux mint 21 mate edition (that is all I tested) there is an executable file that can be started.
Once the image is extracted, the arduino-ide file will start the IDE by clicking on it from the caja file manager or executing it from the command line. 

I have no clue why the 2.x release has chosen to use the file name "arduino-ide" vs  "arduino" like all the 1.x releases did.
But at least now you can click on a file that comes with the package.