arduino/arduino-ide #692: Arduino IDE 2.0-rc2: had troubles finding serial port. Chose COM1, now Portenta doesn't' work.

hpssjellis, 2021-12-15:
I am not a beginner!

Tried IDE 2.0 on Windows with the PortentaH7 and it had troubles finding the serial port. Only listed COM1, which when I checked using the command prompt and the "mode" function was the only available port. Selected it and now nothing works.

Went back to the old IDE and the Portenta is fixated on COM1, so I can't even flash anything to it, presently going to try a Linux computer to see if I can unbreak this cycle. Other Portenta's still work fine.

Nope, even on Linux old Arduino IDE the serial port is not available.

I will check the Version 2.0 IDE again, any suggestions!

per1234, 2021-12-16:
Hi @hpssjellis. Please try this:

1. Open [Windows Device Manager](https://support.microsoft.com/en-us/windows/open-device-manager-a7f2db46-faaf-24f0-8b7b-9e4a6032fc8c).
1. Open the "**Ports (COM & LPT)**" section of the device tree.
1. Take note of the ports that are listed there.
1. Connect the Portenta H7 board to your computer with a USB cable.

Do you see a new port appear under "**Ports (COM & LPT)**"?

hpssjellis, 2021-12-16:
HI @per1234 Thanks for getting me to try again. I tried 4 computers at school even a linux computer and all seemed to say the Portenta was bricked. After a year and a half of using the Portenta daily I had never bricked one before. So I tried again on my home computer this time not double pushing the button into bootloader mode. Just doing like you said. see the com ports before then plug in and view them after (P.S. I use "cmd" prompt on windows then type "mode" which shows the COM ports). Well this time I got a port. From the IDE 2.0 I got a port but it was COM1 which I never see. So with a port I flashed a blank skech and it uploaded then I double pushed to get into bootloader mode and it worked. 

Not sure if all along my mistake was getting into bootloader mode, but nice that it worked now.

Any idea why IDE 2.0 did that to the Portenta, seems like a big issue.

Thanks again for the reply @per1234

closed 2021-12-16 by hpssjellis