arduino/arduino-ide #699: Rename INO file deletes sketch and any .h files within folder IDE2.0.0-rc2

icehousedoo, 2021-12-17:
If you attempt to rename the main INO file of your project and save it back to an existing folder, the folders contents are deleted.
Steps to reproduce:
Open IDE 2.0.0-rc2
- Create new tab called MQQT.h
- Add code etc to MQQT.h
- Create new tab called WIFI.h
- add code etc to WIFI.h
- Save project, create new windows folder in save dialogue, use default file name "sketch_dec17a"
Everything looks normal
- Raname WIFI.h tab to WIFIx.h
	- works ok
Sketch / Show sketch folder shows files as expected inc 4x log files

- Rename sketch_dec17a.ino, select original folder, name file MAIN.ino
Contents of source folder are deleted, no longer have the MQQT.h or WIFI.h files.

per1234, 2021-12-17:
Thanks for your report @icehousedoo. I just gave it a try and was unable to reproduce the problem.

Please provide more detailed instructions that will allow me to exactly replicate the steps you took when you encounter this bug.

If possible, remove any unnecessary steps to facilitate the investigation. Is it really necessary to create two additional tabs? Is it really necessary to rename one of the tabs? Is it really necessary to add code etc to the tabs?

Do you have **File > Preferences > Auto save** enabled or disabled?

icehousedoo, 2021-12-17:
Ah ha my fault, the old PEBKAC issue. 
When renaming the INO file, it is moved to a new folder with matching name, along with any associated TAB files I create.
This is new behaviour I probably would've known about if I'd read the release notes ;-) 
Sorry for wasting your time, I did appreciate your prompt support.

Cheers

closed 2021-12-17 by per1234