arduino/arduino-ide #707: ide2 does still not run on arm linux / offline compilation

abusr, 2021-12-20:
When will you release ide2 for arm linuxes? Are golang or node js tied to only x86/x64? Nope! LLVM or GCC? Nope! Typescript? Nope! So which could be the reason for not officialy releasing ide2 for linux/arm? If it has been released for linux/pc, which could be the everest class obstacle for not have been released for linux/arm in the same manner?

Also, if one tries to build the ide2 piece by piece, (s)he'll need lots of stuff as I'm beginning to see, both yours and external. Fellow devs, can you at least gather all these ingredients in one combined repo with a build script of some sort in front so that - someone could at least try to build your ide offline?

Please consider my thoughts. Thank you in advance.

per1234, 2021-12-20:
Hi @abusr. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/107.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-12-20 by per1234

abusr, 2021-12-20:
(thank you per1234 for pointing out an existing thread for my first paragraph. have you got any reply regarding what I asked? also, please consider the combined repo which could serve as a temporary substitute for the missing release)

When will you release ide2 for arm linuxes? Are golang or node js tied to only x86/x64? Nope! LLVM or GCC? Nope! Typescript? Nope! So which could be the reason for not officialy releasing ide2 for linux/arm? If it has been released for linux/pc, which could be the everest class obstacle for not have been released for linux/arm in the same manner?

Also, if one tries to build the ide2 piece by piece, (s)he'll need lots of stuff as I'm beginning to see, both yours and external. Fellow devs, can you at least gather all these ingredients in one combined repo with a build script of some sort in front so that - someone could at least try to build your ide offline?