arduino/arduino-ide #709: RC2 Building sketch hangs

sterretjeToo, 2021-12-21:
I did try to find this but did not see it.

IDE 1.8.5 is installed as the main IDE, RC2 is installed as a zip. Using Windows 10 home, fully patched.

1) I left the IDE open last night with a sketch for an Uno Wifi Rev2; this sketch was not saved.
2) This morning I copied a (non Uno Wifi Rev2) sketch from the forum and pasted it over the existing sketch; the new sketch is 242 lines and contains a error.
3) I pressed `<ctrl>T`; this might not be relevant.
4) I noticed that the status bar showed "building sketch".
5) More than 30 minutes later, the message was still there; I noticed one message about indexing but that disappeared again.
6) Closed the IDE and restarted it; empty sketch (no problem for me)
7) Pasted the forum sketch again
8) "Building sketch" shows again in the status bar, this time it does not hang and the message inthe status bar disappears.

I would expect the build process of the sketch to complete.

This might just have been a glitch; reported as an observation.

per1234, 2021-12-21:
There is a related report about the "indexing" stage of the language server sketch processing operation hanging:
https://github.com/arduino/arduino-ide/issues/431
I have periodically encountered that one. I don't remember observing a  hang at "building" though, so I think it is worth tracking it as a separate issue for now.

ubidefeo, 2022-04-21:
@cmaglie 
this is certainly due to the LS being killed by the IDE after a long time.

@sterretjeToo
we are still looking for the source of the issue, but essentially a `kill` command is sent tot he language server at unpredictable times.
The criticality of this bug has been upped.

fstasi, 2022-06-08:
this should be fixed with IDE rc7

per1234, 2022-06-09:
@fstasi I looked at the 2.0.0-rc7 release PR (https://github.com/arduino/arduino-ide/pull/1027) and it doesn't appear there is anything in it that could possibly affect this behavior (it only updates the version metadata).

Was it https://github.com/arduino/arduino-ide/pull/1021 that is expected to fix this issue?

qwertychouskie, 2022-06-17:
This could also be related to https://github.com/arduino/arduino-ide/issues/714

per1234, 2022-07-10:
@fstasi please answer the question I asked above.

per1234, 2022-08-24:
Hi @sterretjeToo are you still experiencing this problem?

Some very significant language server improvements have been made since the time of the report (e.g., https://github.com/arduino/arduino-ide/pull/738, https://github.com/arduino/arduino-ide/pull/1019, https://github.com/arduino/arduino-ide/pull/1107, https://github.com/arduino/arduino-ide/pull/1255). It is possible one of those also fixed this issue.

per1234, 2022-10-11:
Hi @sterretjeToo. Please let us know whether you are still experiencing this problem when using the latest version of Arduino IDE.

per1234, 2022-11-26:
Closing due to lack of response.

@sterretjeToo if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2022-11-26 by per1234

sterretjeToo, 2022-11-27:
I do apologise for not responding. For some reason I never realised that there was a question :(

Issue can be/stay closed, happily using 2.0.2 at the moment without this issue.

theticraft, 2024-11-16:
Somehow, I get the same issue in a later version (IDE 2.3.3). After restarting, it is working again. Is there a solution that got somehow destroyed accidentally?