arduino/arduino-ide #710: Port changes on native USB upload when two ports are associated with the selected board name

per1234, 2021-12-21:
## Describe the bug

Some Arduino boards have native USB capability which allows the primary microcontroller to produce a CDC serial port over which sketches are uploaded. The upload process for these boards goes like so:

1. Arduino IDE signals the board to run the bootloader.
1. Sketch program stops, which kills the CDC serial port it produces.
1. Bootloader program starts, which produces another CDC serial port.
1. Upload happens.
1. Bootloader program stops, which kills the CDC serial port it produces.
1. Sketch program starts, which produces another CDC serial port.

During this process, the port disappears and reappears. This causes the IDE to select the wrong port if any other ports have been associated with the currently selected board name .

The obvious way that multiple ports might be associated with the same board name is having multiple boards of the same model connected to your computer. However, it occurs even with a single board simply by selecting the wrong port once.

## To Reproduce

You will need:

- Arduino board with native USB capability (e.g., Leonardo, MKR, Nano 33 BLE)
- An additional serial port on your computer. This could be an internal built-in port (i.e., COM1), a spare Arduino board connected to your computer, or any other device that produces a port (e.g., FTDI adapter).

1. Select the additional serial port from the **Tools > Port** menu in the Arduino IDE.
1. Select your native USB board from the **Tools > Board** menu in the Arduino IDE.
   You have now associated the selected port with this board name.
1. Select the real port of your native USB board from the **Tools > Port** menu in the Arduino IDE.
1. Select **Sketch > Upload** from the Arduino IDE menus.

:bug: The port selection will now switch to the port that was selected in step (1).

**Note:** The above instructions use **Tools > ...** menu paths for the sake of clarity. The issue also occurs when using the board/port selection dropdown menu and the "**Select other board and port...**" dialog.

## Expected behavior

Retain the port I selected.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc2-snapshot.c064673
Date: 2021-12-15T09:32:43.552Z
CLI Version: 0.20.2 [13783819]

## Additional context

The tried and true established approach used by the classic Arduino IDE handles this just fine. So its behavior can serve as a reference for the solution.

---

There are several other issues related to the IDE not correctly handling the port selection:

- https://github.com/arduino/arduino-ide/issues/43
- https://github.com/arduino/arduino-ide/issues/82
- https://github.com/arduino/arduino-ide/issues/180
- https://github.com/arduino/arduino-ide/issues/287

---

Originally reported at https://forum.arduino.cc/t/each-day-a-new-problem/937997

ben-qnimble, 2022-01-13:
Just to add more details on Arduino IDE 1.6.19, if you have two identical devices connected and select the port of one and then it is disconnected, the IDE does not connect to the other port, but has no selected port. Then when the device is plugged back in, it re-selects the originally selected port. This prevents the IDE from changing ports / devices on the user and I believe is the right behavior and the 2.0 IDE should match this.

ben-qnimble, 2022-01-13:
After more testing, I think is issue is slightly broader. Let's say you have two different devices: COM1 which does not have a unique VID/PID associated with an installed board and COM2 which does a VID/PID that matches a board type. If you select COM2 and then COM2 is disconnected, COM1 will be selected. 

If, instead, COM1 has a unique VID/PID associated with a different board, then when COM2 is selected and this disconnected, the IDE leaves the port unselected. And when COM2 is reconnected, the IDE will select it. 

So I believe the issue of Arduino IDE selecting a new (unwanted) port isn't just in the situation where you have multiple boards connected of the same type, but when you have any extra random ports available (say, non Arduino devices), which seems likely to be much more common.

ben-qnimble, 2022-01-22:
Please see https://github.com/arduino/arduino-ide/pull/765 for more details, but that has a partial mitigation for this when connecting to a device with a protocol other than serial.

In the process of looking into that I think I understand better the underlining issue. When a user selects Board A, then when the IDE looks up the different available ports, for any port that is doesn't know (not VID/PID match) it gives it the name/FQBN of the selected board A.  Then when the port disconnects, the IDE tries to find a match from the remaining ports, but a name/FQBN match is good enough and since all unknown ports match this, it will reconnect with any unknown port.

closed 2022-08-24 by davegarthsimpson