arduino/arduino-ide #712: Add Missing VScode translation files for languages translated via Transifex by the community

rsora, 2021-12-22:
In order to consider completed a translation and available to users, we need to merge translations coming from the community via Transifex and available here https://github.com/arduino/arduino-ide/pull/705 and Vscode translation file already available

per1234, 2022-04-08:
This has now been done for the six languages with the highest localization coverage of the Arduino IDE-specific content on [Transifex](https://www.transifex.com/arduino-1/ide2/): https://github.com/arduino/arduino-ide/pull/941

per1234, 2022-04-22:
Since the time of the initial localization work, several additional languages have reached a high coverage state thanks to the excellent contributions from the Arduino community. These should also be added to the Arduino IDE build.