arduino/arduino-ide #714: Indexing/building performance issues

Wezz19, 2021-12-22:
**Describe the bug**
- High CPU and RAM usage for the IDE process, when working on a sketch for longer than 30mins.
- Indexing and building never stops after 60+ mins
- Scrolling the editor is extremely slow and choppy after 60+ mins
- Hovering the pointer over variables never brings up additional info about them, after 30+ mins
- (The IDE has never crashed for me though, even after 4+ h usage)

(all the CPU and RAM usages are for the IDE 2.0 process)
1. (Time since IDE launch: 0mins) CPU 0% and RAM about 400MB. Indexing completes in less than 2s.
2. (Time since IDE launch: 30mins) CPU 47% and RAM constantly jumping between 1100->500MB, indexing and building sketch takes longer and longer.
3. (Time since IDE launch: 60mins) CPU between 57-49% and RAM jumping between 2300->1200MB. Memory graph in Task manager looks like a saw blade. Indexing and building never finishes.
4. (Time since IDE launch: 0mins) Saving and restarting the IDE brings the CPU down to 2% and RAM to 500MB.

**To Reproduce**
1. Load up a sketch
2. Edit, verify and upload for at least 30 to 60mins

**Workaround**
Restart the IDE every 30min

**Hardware/Software**
 - OS: Windows 10
 - IDE: 2.0 nightly 20211219
 - CPU: i5-6200U
 - Board: Nano clone

**Additional context**
Possibly related issues:
https://github.com/arduino/arduino-ide/issues/431
https://github.com/arduino/arduino-ide/issues/709

IanBUK, 2022-01-24:
I'm getting performance issues with RC3 on MacOS that are resolved by restarting the IDE. I'll check to see if memory/cpu usage trends upwards.

Zer0-bit, 2022-01-29:
Getting the same behaviour, seems like it might be related to some HW acceleration feature as when this behaviour starts and i upload a sketch the notification says "Done uploading" as usual while the terminal itself is still outputting compile data and stuff, as in the UI is waaayyyyy behind the actual process.

Ali229, 2022-03-02:
Getting similar behavior, the IDE is extremely slow, and keeps indexing/building and take system resources while doing it, during this time the IDE stops error checking or keeps showing error on code that has no syntax errors. For now it's so slow and unusable that I have to use another IDE to write code, when I'm done I copy it to Arduino and compile!

IanBUK, 2022-03-02:
I’ve gone back to version 1.8.3 of the ide. Yes, I’ve lost all the new features, but at least it doesn’t slow down so much. 

This is a real barrier to release.

IrregularShed, 2022-03-07:
I'm in the exact same situation unfortunately. I don't like using the old IDE because it's not got many of the niceties that I've grown to expect over the past decade with other IDEs and editors (VS Code really spoiled us) so it slows me down; sadly the new one slows me down in another way.

peter76, 2022-04-22:
Same problem here on Fedora linux; editor turns unbearably slow after 50 lines of code or so; just keeps indexing and intellisense stops working. VS Code works like a charm with the arduino extension installed.

Just to sort out some misconfiguration, I added a new user on my system and ran a freshly downloaded ide from there, same problem.

Let me know if I can do something to look into this, the ide looks promising, but is unusable as it is:-(

IanBUK, 2022-04-22:
I’ve migrated to VS Code with PlatformIO. I’ve seen that there’s an Arduino extension too, I might try it over the weekend.

VS Code just works.

peter76, 2022-04-23:
> I’ve migrated to VS Code with PlatformIO. I’ve seen that there’s an Arduino extension too, I might try it over the weekend.
> 
> VS Code just works.

I'm using VS Code with the Arduino extension and the Arduino-cli; took a bit of figuring out, but works perfect. PlatformIO was always a bit too much for my needs.
But still hope the Arduino-ide gets figured out, the Arduino platform deserves a better ide than the old one imo.

IanBUK, 2022-04-23:
I migrated from Windows to Mac last year. Until then I was using Visual Studio with VSMicro. That seemed to work well.

xinkiknix, 2022-04-27:
For me the issue got solved by renaming all directories (also hidden directories) containing any reference to arduino installed items. Uninstalling Arduino still left some directories (for future reuse?). 
I am running version 2.0 rc6, and it seems to behave ever since.

JiriBilek, 2022-04-27:
I always do a clean install (delete the Arduino IDE folder and create a new one) and the problem still persists. 
Using this IDE with ESP32 toolchain is quite annoying. On the other hand, the VSCode runs like a charm, so the problem is in the Arduino IDE and not in the toolchain, I think.

IrregularShed, 2022-04-27:
I got a brand new laptop a couple of months ago and installed this on it as one of the first pieces of software - there was no difference for me either.

serpinio, 2022-05-02:
The same for me, but with ESP32 it takes much less time for the editor to become "stuck" due to the fact that the chip supports WiFi and Bluetooth. Because of that even a blank sketch with WiFi enabled takes 800 KB on flash after compilation. So a **lot** of code is recompiled compiled _every time I make the slightest change in my .ino file_. 

**I have a proposal:** please add _an option to only execute language server after quicksave/manual save_. The biggest issue is that the editor rushes to recompile everything on **each** character type in the editor. I tried disabling auto save, but nothing changed. I believe this is simply wasteful to trigger recompilation on the tiniest change in code. Sure, if the code you changed wasn't verified yet, you won't be able to see your changes in tool-tips and suggestions, but it makes much more sense _to give the programmer the choice when to re-build the code_. Considering that Arduino IDE 2.0 compiles in single-core, seems like it places those indexing/building tasks _in queue_. So if you type, then pause for a couple seconds, then type again and do so for like 10-20 characters, that's it. The language server queue is overwhelmed and you're never getting your suggestions and have to work with a super-sluggish editor.

Adding a switch to only re-index/re-build the sketch on save will also help with editor completely _freezing_ when it's in such an overwhelmed state and you open _another_ project window. For me in 95% of cases this guarantees complete IDE freeze in 5-20 seconds for both windows. Sometimes it's so bad, the OS becomes unresponsive. Not a single other app I use does this.

tl;dr; 

**Please add an option to re-index/re-build the sketch only on save (manual or auto).** This way one could change auto-save to happen only once a minute and the editor would have plenty of time to finish the re-index/re-build process even for monstrously huge tool-chains like the one for the ESP32 chips.

And thank you for a very useful editor which is available for Linux unlike VCode.

StefanL38, 2022-05-27:
[ptillisch](https://forum.arduino.cc/u/ptillisch) asked If I can provide detailed information about my system

I'm using a 16GB RAM 2TB-SSD Lenovo Thinkpad Tablet Gen.2 
not an uptodate HIGH-End gaming machine. But pretty fast. 
I have no performance problems with anything else

Example:  I use the software mailstore Home 
https://www.mailstore.com/de/produkte/mailstore-home/
Which has indexed 8,2 **GB** of email-content
If I let do mailstore home a search inside the 8,2 GB for a keyword the results appear within 2 **seconds**.

Gerätename	X1-Tablet-16-2
Prozessor	Intel(R) Core(TM) i7-7Y75 CPU @ 1.30GHz   1.61 GHz
Installierter RAM	16,0 GB (15,8 GB verwendbar)
Geräte-ID	CDE53F97-A7E0-43E3-A7EE-C3FD33BA0B61
Produkt-ID	00330-80000-00000-AA986
Systemtyp	64-Bit-Betriebssystem, x64-basierter Prozessor
Stift- und Toucheingabe	Unterstützung der Stift- und Toucheingabe mit 10 Touchpunkten

Edition	Windows 10 Pro
Version	21H2
Installiert am	‎27.‎03.‎2021
Betriebssystembuild	19044.1706
Leistung	Windows Feature Experience Pack 120.2212.4170.0

I have installed additional boards
ESP8266, ESP32, Seeeduino SAMD, PJCR Teensy
I also attach the filelist of Arduino15 and Arduino mainfolder

[Arduino15-Dirlist.zip](https://github.com/arduino/arduino-ide/files/8786148/Arduino15-Dirlist.zip)
[Arduino-main-Dirlist.zip](https://github.com/arduino/arduino-ide/files/8786156/Arduino-main-Dirlist.zip)

Both have about 40.000 lines = files

I'm using F-Secure 2022 as the internetsecurity suite. 
I have quite some basic knowledge about Windows so if you can give me some hints what kind of information would be helpful to reproduce the issue then post this information

You should make **Stop Language Server the default**

    Press CTRL+SHIFT+P
    Type "server" in the search field
    Click "Arduino: Stop Language Server"
    Enjoy working in a super-responsive IDE with lightning-fast suggestions
    When ready to re-index, select the "Arduino: Start Language Server" command and wait for a re-index

quote of user ptillisch at Arduino.cc

> Even though I think there is surely some room for improvement in the results we are experiencing, none of the members of the team working on the Arduino IDE 2.x are able to reproduce the exceptionally long processing times reported by those users. This makes the problem difficult to investigate, and validation of any potential improvements or fixes difficult to validate.

This makes me think:
maybe you are doing the same bad habits as microsoft-developpers do
Using superhigh-end computers with 64GB of RAM 32 GB-RAM-disk and top-speed SSD-Raids 
for developping. This is OK to speed up the rebuilding process 
but 
when it comes to test the application-performance you should use a 3third generation processor in a 4GB-RAM system using a classcial HDD with a lot of additional boards installed.

IanBUK, 2022-05-27:
I've come up with a good work-round on my MacBook Air: use VSCode.

hav00k, 2022-06-02:
I'm also experiencing this issue using a fresh install of 2.0.0-rc6.

When I start typing some new code it seems the "parser lags behind", as I'm standing there observing how the "wrong syntax" red lines appear and disappear with a considerable lag behind of what I've just typed. During this time the indexing/building process never stops.

It comes to a point where this indexing/building just goes forever, and I'm unable to use the auto-format feature, the suggestions, nor the go to functions (by my understanding these depend on the Language Server). The only solution is to restart the IDE, which fixes it.
In the current state I basically have to do this after typing every other line.

My specs (desktop machine):
Ryzen 1800X (8-core, 16-thread) @ 3600Mhz
32GB DDR4 RAM @ 3200Mhz
All SSD (Samsung)
Windows 10 Pro 21H2 (build 19044.1706)

qwertychouskie, 2022-06-16:
Affects me too.  Ryzen 5900HS, 16GB RAM, Ubuntu 22.04.  Editor becomes completely unusable quite quickly.

I find it quite concerning that the devs haven't commented on this issue at all, considering it makes the IDE completely unusable out-of-the-box.  I'm going to follow @serpinio's suggestion for now and completely disable the language server, but I'm hoping the issue is fixed soon.

boundmaidlea, 2022-06-19:
VSCode has an option, that this app seems to miss: `C_Cpp.intelliSenseUpdateDelay` makes it, so the IDE does not attempt to reprocess the code, and instead waits a custom amount of milliseconds, default 2000, which highly improves performance just by itself. Currently i can see the parser rereading the script for each of my typed in characters separately, as when i hover over a newly typed variable name, the variable's name appears letter by letter in the "Use of undeclared identifier: '<variable name>' clang(undeclared_var_use)" tooltip, basically as one letter per second. Same behaviour for the red squiggly lines below.

StefanL38, 2022-06-19:
Hi boundmaidlea,

good catch if this is causing the performance issue. Only in case it is  just a tiny effort you:
Can you explain why this thing did **not** affect the developpers in the same way?

Could it be that the developpers are working in a local GigaBit-Network with everything local available with 0,0x milliseconds ping-times and high performance intelligent GigaBit network-switches?

If Yes:  When will these developpers finally learn to develop under real average conditions?
The annyoing of having to wait another five seconds for the new compile-run to be finished would educate them to 
a different style of developping and point them to performance-problems right away.

Just my 0.2 cents of not knowing how it really works.

FastThenLeft, 2022-06-24:
I have this same issue with Arduino 22.04 and IDE rc8. The large increase in CPU usage seems to correspond with the following error in the log. This error loops forever putting the same info over and over in the log.

2022-06-23 13:24:04 root ERROR Error: Aborted onWillSaveTextDocument-event after 1750ms
    at file:///tmp/.mount_arduinfquyjl/resources/app/lib/bundle.js:2:6680353
2022-06-23 13:24:08 root ERROR Error: Aborted onWillSaveTextDocument-event after 1750ms
    at file:///tmp/.mount_arduinfquyjl/resources/app/lib/bundle.js:2:6680353
2022-06-23 13:24:10 root INFO isTemp?
2022-06-23 13:24:10 root ERROR Uncaught Exception: 
2022-06-23 13:24:10 root ERROR Error: Connection got disposed.
    at Object.dispose (/tmp/.mount_arduinfquyjl/resources/app/node_modules/vscode-jsonrpc/lib/main.js:904:25)
    at /tmp/.mount_arduinfquyjl/resources/app/node_modules/vscode-ws-jsonrpc/lib/socket/connection.js:14:41
    at CallbackList.invoke (/tmp/.mount_arduinfquyjl/resources/app/node_modules/vscode-jsonrpc/lib/events.js:62:39)
    at Emitter.fire (/tmp/.mount_arduinfquyjl/resources/app/node_modules/vscode-jsonrpc/lib/events.js:121:36)
    at closeHandler (/tmp/.mount_arduinfquyjl/resources/app/node_modules/vscode-jsonrpc/lib/main.js:240:26)
    at CallbackList.invoke (/tmp/.mount_arduinfquyjl/resources/app/node_modules/vscode-jsonrpc/lib/events.js:62:39)
    at Emitter.fire (/tmp/.mount_arduinfquyjl/resources/app/node_modules/vscode-jsonrpc/lib/events.js:121:36)
    at WebSocketMessageReader.fireClose (/tmp/.mount_arduinfquyjl/resources/app/node_modules/vscode-jsonrpc/lib/messageReader.js:111:27)
    at WebSocketMessageReader.fireClose (/tmp/.mount_arduinfquyjl/resources/app/node_modules/vscode-ws-jsonrpc/lib/socket/reader.js:67:19)
    at /tmp/.mount_arduinfquyjl/resources/app/node_modules/vscode-ws-jsonrpc/lib/socket/reader.js:24:18

theo222, 2022-06-29:
I see the same problems here, especially when coding for esp32.
After a (short) while, the IDE says it's indexing or building something, while i am typing.
I can even hear it on my computer's fan that it's working hard (i5-9400).
Code insight says "loading" for minutes.
Working is getting uncomfortable this way.
A restart helps for a short time.

FastThenLeft, 2022-06-29:
Yes, this is a BIG problem. I teach microcontroller programming with Arduinos at the local community college. I've been hoping to switch to the new IDE so yesterday I tried rc8 on their older i7 computers. Within 5-10 minutes the CPU usage was about 60% and the fan was going full blast. The IDE became very sluggish.

MikesMachines, 2022-06-30:
Running RC7 on my dev workstation (i9, 32GB RAM), the IDE has been chewing up resources. It indexes 100% of the time. I uninstalled everything and reinstalled with a portable setup, moving all folders (including libraries) into the portable folder. That didn't help. I disabled Win Defender, which did help a little, but the IDE was still using way too much CPU and RAM.

**Solution - Stopping the language server, as @serpinio wrote dropped the CPU usage from 47% to 0.3% and the memory usage from ~7GB to 480MB.** 

Thank you @serpinio  !!!

theo222, 2022-07-01:
@MikesMachines : Yes, it's a great workaround for now, but it's not a **solution**.

StefanL38, 2022-07-01:
After reading the comments I have a question:

Did the developping-Team test the IDE 2.0 with **activated** language server ?
If Yes:
Did the developping-Team test the IDE 2.0 with activated language server that has a configuration as the configuration will be for the average user? 

With "the configuration will be for the average user I mean"

This server is a server that is a **distant** server with a bandwith of 50 MBit ?

@Development-Team:  I highly doubt it that you tested the IDE 2.0 with activated language-server at all !
If you did You should **hurry up** to find out what special configuration you configured to make the problem vanish!

You should invite a **real newcomer** to make an installation of Arduino IDE 2.0 all on her/his own on a freshly installed Windows 10 computer letting her/him do all steps on her/his own but very carfully watching and writing down what this newcomer does to be able to compare it with your way of installing and testing it.

**Man !** 
With each day more users were **mis**used as the alpha-testers (not beta-testers !) the image of the IDE 2.0  becomes worse and worser as the new  benchmarking for highly agressive  CPU-usage and RAM-grapping !

Making a dev workstation (i9, 32GB RAM), CPU usage 47% and memory usage ~7GB 
This is far off from any average user !

qwertychouskie, 2022-07-01:
@StefanL38 Internet speed has nothing to do with it, the language server doesn't refer to an Internet server, but rather a compilation/indexing/etc process ran by the IDE.

per1234, 2022-07-01:
@StefanL38 we have different standards of behavior on Arduino's GitHub repositories than on the forum. 

This is a place for people to work productively on very complex software projects. For that reason, all activity here must be very tightly focused.

If you have concrete information to share about the problem, you are welcome to post here, but evidence-free speculation and whining is not appropriate here. You are always welcome to start that sort of discussion on the Arduino Forum, and I will be happy to participate in it there.

StefanL38, 2022-07-01:
Hi per1234 thank you for clearing this. Which Arduino-**sub**forum is the right one for this kind of discussion?
And additional are the **developping**-experts reading this Arduino-Forum too? To me it seems that the developping-experts are paying no attenttion to this problem.

IanBUK, 2022-07-01:
Might it be worth a project dev having a session with someone experiencing the issue, a Zoom call with screen sharing?

Perhaps that could lead to some extra investigation or diagnostics which uncover the root cause and point to a solution.

Apologies if this has already been tried.

xinkiknix, 2022-07-19:
> completed via 57841b3

The real problem is not yet solved.
I am running nightly build 19/07 and indexing is still running like crazy

[image: image.png]

Groeten/Greetings

Johan terryn

kittaakos, 2022-07-19:
Thank you for trying out the nightly build and reporting us back.

per1234, 2022-07-28:
Hi all. We would like to better pinpoint which components are involved with this problem.

The problem seems to be related to the [Arduino Language Server](https://github.com/arduino/arduino-language-server), which is not feasible to use directly since it is designed to be integrated into an IDE or editor. What we can do to easily isolate the language server from the other components of the Arduino IDE is use it from an alternative editor, then check whether the problem followed the language server or not.

For this purpose, @kittaakos has prepared a minimal [Visual Studio Code](https://code.visualstudio.com/) extension that runs the Arduino Language Server. It will be very helpful if you would install this extension, use it on a sketch, and then report here whether this problem of extraordinary resource usage still occurs.

Instructions for that are located here in the associated discussion thread on the Arduino Forum:

https://forum.arduino.cc/t/cpu-utilization-very-high/941187/58

Please let me know if you have any questions or problems while following them.

xinkiknix, 2022-07-28:
I did this as requested:
Besides that the sketch (any sketch) can not find required libraries (e.g. Arduino.h) the CPU utilisation is very low
 result:
[image]

kittaakos, 2022-07-29:
I appreciate your help, @xinkiknix ❤️ 

Could you please tell me how did you try to compile the sketch? What did you do?

I pushed two minor fixes for the VS Code extension (see the commit ref [here](https://github.com/arduino/vscode-arduino-tools/commit/85d2f53eee7f3a7ae180ad7985ee8fa586ab0b3c)). Could you please uninstall the test VS Code extension (`vscode-arduino-tools-test`) and use the same download link to re-download and re-install it again? The link did not change: https://github.com/arduino/vscode-arduino-tools/raw/run-arduino-ls-in-code/build-artifacts/vscode-arduino-tools-test-0.0.2-beta.4.vsix

Thank you!

xinkiknix, 2022-07-29:
I assumed it tried to compile while running the Arduino:verify?.

I installed the latest as requested and ran some test, changing code, adding comments, ... and running start language server and verify.

While running start language server I notice that endpoint security starts eating all CPU
[image]
Stopping anti-virus reduces the time required to run the process.

Same I observed in latest nightly build, any code change results in endpoit going to +80% for a very long time.

while running verify  clangd.exe barely goes beyond 8% short peak of 28%. Endpoint does not show up as using much CPU.

Is there a way to make the AV software skip checking while updating local files from within Arduino?

InstantMuffin, 2022-07-30:
> Stopping anti-virus reduces the time required to run the process.

Yes. It is unrelated to the issue at hand, but the language server does write the compiled data to the hard drive which on-access AV software generally checks.

> Is there a way to make the AV software skip checking while updating local files from within Arduino?

If you like to use AV and it's on-access scanning feature, you can exempt specific directories from the scan. Unfortunately I can't tell you where the cache is built on windows. I would suspects either in the temp folder or somewhere in the appdata directory structure.
On XUbuntu the cache looks like this, feel free to use filenames and foldernames (exempting the number in the root folder's name) to do a filesystem search on your system:
[image]

xinkiknix, 2022-07-30:
This has no effect on AV utilisation, there must be other folders where data is written. I excluded the complete temp directory...
In the nightly build 0729 I see little CPU utilisation after a while, <3% while Building Sketch is still running...
[image]
and this runs for a very long time, eventually stops. After restarting the IDE CPU utilisation goes to 0.1%

InstantMuffin, 2022-07-30:
So as an affected user running linux, apart from the vscode test which I have done (I could not reproduce the issue there so far), what can I do?

-I tried running the Arduino IDE's profiler. It doesn't work on linux and indefinitely hangs on "processing" after stopping the recording (I waited for the "processing" to finish for a couple of hours to no avail)
-@per1234 mentioned there is a debugger as part of Arduino IDE's featureset in the developer tools, I assume it is under "sources"?
-Reproduce this in a VM and send that VM+snapshot to a dev?
-Let a dev use TeamViewer etc. when the IDE encounters the bug? Any tools you would like me to install beforehand?

xinkiknix, 2022-08-01:
I did run the test on Friday, and did some testing during the weekend and today (running nightly build).
The VS Code "**Arduino: Verify**" runs fine now, no errors and no extensive CPU utilisation, only short bursts when running "**Arduino: Verify**". When switching boards ( "**Arduino: start language server**" I see a significant difference between boards. And I think I found the reason why tasks run slow and high CPU utilisation:
`"**CC1plus.exe**"` and "**xtensa-esp32<xx>-elf-g++.exe**" for the respective boards are not signed, this triggers AV and runs CPU to 100% on all cores. I excluded these files in the AV for the boards I use (ESP32 & ESP32-S3) for these boards both "**Arduino: Verify**" and "**Arduino: start language server**"  run very fast and with little CPU utilisation. 
In Arduino nightly build the exclusion of these files in AV make the IDE run way much faster for verify and compile/Upload. the difference is from compile time of over 5 min. to compile < 1min, upload similar. 
None of the Arduino components run with high CPU utilisation most of the time. There was one instance where 1 process ran for a few minutes with high CPU, but showing no activity in the IDE. (no Build, no Indexing,...). This process had as commandline:

`>     11164 "D:\NightlybuildArduino\Arduino IDE.exe" --type=renderer --user-data-dir="C:\Users\Johan\AppData\Roaming\arduino-ide" --app-path="D:\NightlybuildArduino\resources\app" --no-sandbox --no-zygote --field-trial-handle=1588,2066934688397159316,1515606472287176484,131072 --disable-features=SpareRendererForSitePerProcess,WinRetrieveSuggestionsOnlyOnDemand --lang=en-GB --device-scale-factor=1 --num-raster-threads=4 --enable-main-frame-before-activation --renderer-client-id=4 --no-v8-untrusted-code-mitigations --mojo-platform-channel-handle=2516 /prefetch:1`

That is it for today...

InstantMuffin, 2022-08-02:
I think this may be helpful.
I just booted up the IDE and autosuggestions worked fine, and there was no load.
I could see "indexing" in the bottom left corner. I'm not sure if this happened at one of the I assume files it was indexing (at a specific count) but as soon as it was done the load went up to 150% CPU.
It worked fine throughout at least part of the indexing process, then suddenly nothing and high load.

xinkiknix, 2022-08-02:
I have been testing the latest nightly build (02/08) and am satisfied with the behaviour, Things run smoothly with very low CPU usage.  I still have CC1plus.exe and xtensa-esp32<xx>-elf-g++.exe for the respective boards (as they are not digitally signed) as exceptions in my AV. 

For me this looks it behaves as one would expect.

InstantMuffin, 2022-08-03:
I have been testing rc9.1 for a short while now. I usually expect slowdowns after less than 10 minutes of use.
It has been more than half an hour, with much more time running the IDE in the background.

I will be using Arduino IDE during the next days and will report back if any slowdowns occur.
But honestly as far as this looks and feels to me (so odd to see how many features the IDE has now that slowdowns don't cut them out) this is fixed.

If you don't hear back from me until Friday, this is completely gone on my end.

kittaakos, 2022-08-04:
I would like to say thank you to all who helped us with the verification and testing of the Arduino language features. You're great. 🙏 

I will keep this issue open for some time and will close later when no critical issues pop up.

----

>  language server does write the compiled data to the hard drive which on-access AV software generally checks.

This is done by the [Arduino CLI](https://github.com/arduino/arduino-cli).

> Unfortunately I can't tell you where the cache is built on windows. I would suspect either in the temp folder or somewhere in the appdata directory structure.

Here is the [code](https://github.com/arduino/arduino-cli/blob/2dd8976683f4a9f83fcee478620f1e83d11111b2/arduino/sketch/sketch.go#L315-L323).

An example URI of a temp folder
 - on macOS:
    ```
    file:///private/var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/arduino-sketch-7D2D66808EB316067632A2C48E34D662
    ```
 - on Windows:
    ```
    file:///c%3A/Users/kittaakos/AppData/Local/Temp/arduino-sketch-7320986B49B567949951B77342A53F22
    ```

⚠️ Please perform any antivirus adjustments at your own risk.

Wezz19, 2022-08-05:
Indeed, the latest nightly (20220805) runs very well for me now also. I've been testing it for a few hours today, on the same weak laptop as in December.

Memory usage for the IDE always stays below 500MB and CPU under 5%. A huge improvement!

@kittaakos your latest fixes seems to have been a success! May I ask what exactly in this [PR](https://github.com/arduino/arduino-ide/pull/1255) was the problem? Did some task queue up for each keypress?

By the way, is it possible to manually trigger an indexing without doing a verify? Sometimes they doesn't seem to trigger.

aliphys, 2022-08-09:
I have been using the nightly version, and it has been stable when it comes to CPU/RAM usage. Thanks! cc @per1234 
https://github.com/arduino/arduino-ide/issues/1240

Micahfriesen, 2022-08-09:
Everything is very fast now, great work!

InstantMuffin, 2022-08-09:
After almost a week with hours of use I can now definitely confirm that _any_ performance issue is gone for me!

closed 2022-08-10 by kittaakos

DaleSchultz, 2022-09-20:
well, I just installed IDE 2 official release

**Editor is unusable**, just clicking somewhere takes about 1 second for the cursor to appear! Typing some text (even comment text) has a terrible lag and some characters even get lost.
Ran IDE 1.8.16 and it works fine, responsive.
I have not yet done any folder exclusion in the AV or turned language handling off.

If I create a new sketch and type junk into a comment I see Indexing n/85 alternating with "Building sketch" and this is which essentially an empty sketch.

Task manager shows IDE using 660Mb of memory with an empty sketch.
System:
Device name	-------
Processor	Intel(R) Core(TM) i3-10105 CPU @ 3.70GHz   3.70 GHz
Installed RAM	8.00 GB (7.74 GB usable)
Device ID	----------------------------
Product ID	------------
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

Edition	Windows 11 Home
Version	21H2
Installed on	‎12/‎2/‎2021
OS build	22000.978
Experience	Windows Feature Experience Pack 1000.22000.978.0

DaleSchultz, 2022-09-20:
Today I excluded the temp file folder where the temp files are written and that made the IDE responsive again.  We absolutely need an option to reindex on demand rather than on every keystroke. Most users are not going to be able to find this thread and parse out what has to be done to make the IDE usable.

StefanL38, 2022-09-20:
I'm guessing all the fixes made here in the pre-release-version did
NOT find  their way into the "official release"
@Arduino development-team: did you ever notice that you are no longer in a small village and private garage-team just annoying some students of the same class with amateur errors? You are in a **worldwide public**.

Yes all this is voluntary and for free.  This means there is absolutely **no** pressure for a certain release-date. Go on testing **un**-released beta-testing until the more special systems like DaleSchultz ones reports "runs flawlessly"

per1234, 2022-09-20:
@StefanL38 I already explained to you that this sort of evidence free wild speculation is not appropriate here.

The code is open for anyone to see exactly what is in the release. If you have clear evidence that a fix somehow went missing, then you can provide that evidence. If you don't have that evidence, then you would be better to keep quiet because you only waste all of our time and make yourself look the fool.

JiriBilek, 2022-09-20:
For me the Arduino IDE 2.0 works fine. No lagging. I am using it with ESP32 toolchain.

[image]

FastThenLeft, 2022-09-20:
As a data point, I had the original problem with performance issues the more I used it. But, since rc9.1, every release has been running fast using minimal system resources on my 11-year-old laptop. I also have an entire college class using V2.0.0 with no performance issues.

xinkiknix, 2022-09-21:
I am happy with version 2.0.0 no (major) issues, excellent performance. Using multiple instances with loads of tabs and no performance impact. Fit for purpose!!!

CreativeRobotics, 2025-10-02:
Is this issue ever going to get fixed?

commit abe6a044 (2022-07-11): Notify the LS about the new `build_path` after verify.

Closes #714
commit 0a42fa5d (2022-07-11): Notify the LS about the new `build_path` after verify.

Closes #714
commit a7c85b70 (2022-07-11): Notify the LS about the new `build_path` after verify.

Closes #714
commit 8d462f29 (2022-07-12): Notify the LS about the new `build_path` after verify.

Closes #714