arduino/arduino-ide #716: Library manager list search incomplete

sellensr, 2021-12-22:
**Describe the bug**
Library manager doesn't return a list of all matching libraries

**To Reproduce**
1. Go to Library Manager
2. type "Time" in the search bar
3. Scroll down to find Time by Michael Margolis
4. See that it is not listed
5. type "Margolis" and get no results at all
6. type "timekeeping" or "GPS" and the library is found

**Expected behavior**
Library should be listed under all keywords

**Desktop (please complete the following information):**
 - OS: Mac Monterey and Big Sur
 - Version: 2.0.0 rc2 and rc3

per1234, 2022-06-23:
Thanks so much for reporting this serious bug @sellensr.

Even though this report came first, there is now a duplicate issue that provides a bit more information on the subject: https://github.com/arduino/arduino-ide/issues/1106

Thanks for taking the time to submit an issue.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

per1234, 2022-08-30:
After looking more closely at https://github.com/arduino/arduino-ide/issues/1106, I realized that, although closely related, it was not really a duplicate after all.

This issue is about the search results being incomplete, which will be fixed by https://github.com/arduino/arduino-ide/pull/1361

https://github.com/arduino/arduino-ide/issues/1106 is more focused on a proposal to sort the Library Manager results by relevance. That sorting would have mitigated the problem of the results being incomplete, since it would have made it more likely the library searched for was in the results, but it also is useful even after Library Manager is fixed to show all the results.

closed 2022-08-31 by kittaakos