arduino/arduino-ide #719: arduino.png not found

butor32, 2021-12-26:
**arduino.png not found into ide 2.**

per1234, 2021-12-26:
Hi @butor32. Thanks for your report. We will need a lot more information to investigate this.

Is this an error message you got? If so, where did you see it? What were you doing when you got it?

Which version of the Arduino IDE are you using? It is shown at the top of the window and in **Help > Arduino IDE**.

Which operating system are you using?

butor32, 2021-12-26:
I use Linux (Debian 11.2)  a find -iname "arduino.png" give no return with (arduino-ide_2.0.0-rc2_Linux_64bit)

per1234, 2021-12-26:
What is `arduino.png`?

Why are you looking for it?

butor32, 2021-12-26:
For making icon on desktop, arduino.png are a picture file you can found in arduino 1.8 on lib/arduino.png.

per1234, 2021-12-26:
Here they are:
https://github.com/arduino/arduino-ide/tree/main/electron/build/resources

closed 2021-12-26 by per1234

butor32, 2021-12-27:
Cool to find a icon you install the program and after to have icon you install the git version, I can also install windows version to have a full version.
If you give a version for Linux you must have a icon into the package. 
"Conclusion invalid" F...
You have looking into your github source ? In french we say "Quel bordel"

per1234, 2021-12-27:
We are already tracking the need for an icon on Linux here: https://github.com/arduino/arduino-ide/issues/131 and there is a related issue here: https://github.com/arduino/arduino-ide/pull/356