arduino/arduino-ide #720: yarn start fails: module did not self-register

ben-qnimble, 2021-12-26:
**Describe the bug**
Cannot run 'yarn start' per the BUILDING.md instructions

**To Reproduce**
Steps to reproduce the behavior:
1. git clone https://github.com/arduino/arduino-ide.git
2. cd arduino-ide
2. yarn
3. yarn rebuild:electron
4. yarn start

Output is
```
~/git/testbuild$ yarn start
yarn run v1.22.17
$ yarn --cwd ./electron-app start
$ theia start --plugins=local-dir:../plugins
Starting backend process. PID: 300750
/home/user/git/testbuild/node_modules/bindings/bindings.js:121
        throw e;
        ^

Error: Module did not self-register: '/home/user/git/testbuild/node_modules/drivelist/build/Release/drivelist.node'.
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1034:18)
    at Object.func [as .node] (electron/js2c/asar.js:140:31)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module._load (electron/js2c/asar.js:769:28)
    at Module.require (internal/modules/cjs/loader.js:852:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at bindings (/home/bluey/git/testbuild/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/home/bluey/git/testbuild/node_modules/drivelist/js/index.js:25:27)

```

**Expected behavior**
IDE to run.

**Desktop (please complete the following information):**
 - OS: Linux (Ubuntu 20.04 64bit) and Windows 10

**Additional Info**
I read the Theia IDE prerequisites  documentation and installed node 12.22.8 and yarn 1.22.17 and I'm able to build the Theia example code. Additionally, I tried following the instructions in  
https://github.com/arduino/arduino-ide/blob/cb51966d2392005e3489d360328a28280e373032/BUILDING.md (Pull #706 ) but step 3 
```
cd electron-app
yarn rebuild:browser
cd ..
```
says command not found. I can run it in the root directory, but it makes no difference.

Also, this error looked similar to https://github.com/electron/electron-rebuild/pull/888 and https://github.com/eclipse-theia/theia/pull/10429 so I tried switching to their 1.20.0 from 1.19.0 as that has the work-around pull request integrated, but I get the same error.

WheatleyTheCore, 2022-01-04:
I'm also running into this! I've tried the same steps as @ben-qnimble on an Arch Linux (64 bitinstallation (both using node 12.14.1 with npm 6.13.4 and node 12.22.8 with npm 8.3.0), as well as a fresh Ubuntu install (21.10) in a VM using node 12.14.1.

I've also tried removing node_modules and reinstalling, and even rebuilding everything with `npm rebuild`, neither of which seemed to do anything. 

On both systems I'm able to build and run Theia separately, following the instructions here: https://github.com/eclipse-theia/theia/blob/master/doc/Developing.md#quick-start.  

I've also had my dad (who's much more experienced than I am) give it a go on MacOS, and he's running into the same error. 

Looking a little further back in the error log seems to suggest there's something up with cjs or filenamify (or both). Hope this helps!

fstasi, 2022-01-13:
Hi! we recently updated the build instruction in the main branch. You basically have to rebuild the browser and the electron part of the application in order for the modules to register correctly. Let me know if this fixes your issues

ben-qnimble, 2022-01-13:
I following the updated instructions, but I get an error on step 3. The instructions in [Building.md](https://github.com/arduino/arduino-ide/blob/main/BUILDING.md)

**3. Rebuild the electron dependencies**

```
cd electron-app
yarn rebuild:browser
cd ..
```
but when I do that, I get
```
$ cd electron-app/
$ yarn rebuild:browser
yarn run v1.22.5
error Command "rebuild:browser" not found.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```
If I ignore the error and go to step #4 (yarn start), I get the same error as previously.

fstasi, 2022-01-14:
There was a typo in the readme, thanks for pointing that out.

The right command is

```
cd electron-app/

yarn rebuild:electron

```

I updated the documentation accordingly.

Thanks again!

-- 

Francesco Stasi

Sr Specialist Software Engineer

+39 3407859852

ARDUINO.CC <https://www.arduino.cc/>

| | | | | | | | | | | | | | | | | | | | | | | | | | | | | | | | | | | | | |
| | | | | | | | | | |

This message is confidential and may be legally privileged or otherwise
protected from disclosure;

you must not copy or disclose the contents of this message or any
attachment to any other person.

fstasi, 2022-01-14:
There was a typo in the readme, thanks for pointing that out.

The right command is

```
cd electron-app/
yarn rebuild:electron
```

I updated the documentation accordingly.

Thanks again!

ben-qnimble, 2022-01-14:
Actually, using the new command gave the same error:

```
$ cd electron-app/
$ yarn rebuild:electron
yarn run v1.22.5
error Command "rebuild:electron" not found.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

However, if I run it in the root directory and not in the electron-app directory, then it runs:

```
$ yarn rebuild:electron
yarn run v1.22.5
$ theia rebuild:electron
Processed "drivelist"
Processed "native-keymap"
Processed "find-git-repositories"
Processed "@theia/node-pty"
Processed "nsfw"
√ Rebuild Complete
Done in 12.97s.
```

Unfortunately, I'm still getting the same error on 'yarn start':

```
$ yarn start
yarn run v1.22.5
$ yarn --cwd ./electron-app start
$ theia start --plugins=local-dir:./plugins

Starting backend process. PID: 24432
C:\arduino-ide\node_modules\bindings\bindings.js:121
        throw e;
        ^

Error: Module did not self-register: '\\?\C:\arduino-ide\node_modules\drivelist\build\Release\d
rivelist.node'.
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1034:18)
    at Object.func [as .node] (electron/js2c/asar.js:140:31)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module._load (electron/js2c/asar.js:769:28)
    at Module.require (internal/modules/cjs/loader.js:852:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at bindings (C:\arduino-ide\node_modules\bindings\bindings.js:112:48)
    at Object.<anonymous> (C:\arduino-ide\node_modules\drivelist\js\index.js:25:27)
```

fstasi, 2022-01-17:
@ben-qnimble can you try with this command?

```
cd electron-app/
yarn theia rebuild:electron
```

Thanks

ben-qnimble, 2022-01-20:
Yup, that totally worked! 

```
$ yarn theia rebuild:electron
yarn run v1.22.5
$ arduino-ide\node_modules\.bin\theia rebuild:electron
Processed "drivelist"
Processed "find-git-repositories"
Processed "native-keymap"
Processed "@theia/node-pty"
Processed "nsfw"
√ Rebuild Complete
Done in 59.48s.

$ yarn start
yarn run v1.22.5
$ theia start --plugins=local-dir:../plugins

Starting backend process. PID: 8340

```

Thanks!

fstasi, 2022-01-21:
Based on your feedback, I updated the documentation.

Thanks again for your contribution

closed 2022-01-21 by fstasi