arduino/arduino-ide #721: Incorrect dialog when library dependency already installed

per1234, 2021-12-26:
## Describe the bug

Arduino library authors may specify library dependencies in the `depends` field of [the `library.properties` metadata file](https://arduino.github.io/arduino-cli/dev/library-specification/#libraryproperties-file-format). When using Library Manager to install a library that specifies dependencies which are not already installed, the user is offered the installation of the dependencies.

:bug: This dialog is currently presented even when the dependencies are already installed.

## To Reproduce

1. Use Library Manager to install the "ArduinoRS485" library.
1. Use Library Manager to install the "ArduinoModbus" library.
   (this library [specifies a dependency on the "ArduinoRS485" library](https://github.com/arduino-libraries/ArduinoModbus/blob/1.0.6/library.properties#L11))
   :bug: Notice that a dialog appears which erroneously claims:
   > The library ArduinoModbus:1.0.5 needs another dependency currently not installed:
   > - ArduinoRS485
   > Would you like to install the missing dependency?

[image]

## Expected behavior

Dependencies installation dialog only appears when dependencies are not already installed.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc3
Date: 2021-12-22T15:46:56.004Z
CLI Version: 0.20.2 [13783819]

## Additional context

The issue occurs even after a restart, so it is not dependent on the dependency being installed in the same session as the dependent library, as in the demo.

The issue does not occur with `arduino-cli lib install ArduinoModbus`

ubidefeo, 2022-02-16:
@per1234 
does IDE 1.8.x ignore the ones already installed?

per1234, 2022-02-16:
>does IDE 1.8.x ignore the ones already installed?

Correct. Arduino IDE 1.x only offers to install the dependencies that are not already installed.

That includes the situation where a library has multiple dependencies, some of which are installed, and others not installed. In this case, it will show the dialog, but only list the ones that are not already installed.

ropg, 2025-01-12:
Given that it's been over 3 years, this should maybe be closed with "Won't fix". but that seems silly given that it's quite confusing to users of all libraries with dependencies and that it's such a glaringly wrong UI-thing to happen.

Alternatively to fixing the logic, one could change the sentence on the dialog to something like:

> The library XXXX needs other library/libraries:
>
> YYY
> ZZZ
>
> To install these libraries if they are not installed yet, select "Install All".  

I do C/C++, Python, Lua, Javascript/Typescript (and PHP if you really force me) but I'm not into Java, so I apologize I can't help further by fixing it myself...