arduino/arduino-ide #722: "goto" features don't work for files in sketch `src` folder

per1234, 2021-12-27:
## Describe the bug

Arduino sketches may contain additional code files under [the `src` subfolder](https://arduino.github.io/arduino-cli/dev/sketch-specification/#src-subfolder). This folder is included in the build and source files it contains are recursively compiled.

The Arduino IDE's language server-based features include "Go to definition" and "Peek", which reveal the definition of a code component. This works even for definitions in files not currently open in the IDE, such as libraries.

:bug: The "goto" features do not work for code components from files under the `src` subfolder of the sketch.

## To Reproduce

1. Open this sketch in the Arduino IDE: [HasSrc.zip](https://github.com/arduino/arduino-ide/files/7777754/HasSrc.zip)
1. Select **Tools > Board > Arduino AVR Boards > Arduino Uno** from the Arduino IDE menus.
1. Right click on `foo.h` in this line:
   ```cpp
   #include "src/foo.h"
   ```
1. Select "**Go to Definition**" from the context menu.
   :bug: Note that the file is not opened.
1. Right click on `foo.h` again.
1. Select "**Peek**" from the context menu.
1. Select "**Peek Definition**" from the submenu.
   :bug: Note that the file is not opened.

## Expected behavior

The same behavior from the "goto" features when used on code components from the `src` subfolder of the sketch as from files from any other location.

## Desktop

- OS: Windows 10
- IDE Version: 2.0.0-rc3
Date: 2021-12-22T15:46:56.004Z
- Arduino CLI Version: 0.20.2 [13783819]
- Arduino Language Server Version: https://github.com/arduino/arduino-language-server/commit/277add142bf8d2a98dd2814d79d00ceb0d526d3d
- clangd Version: 13.0.0

## Additional context

Logs: [HasSrc-logs.zip](https://github.com/arduino/arduino-ide/files/7777753/HasSrc-logs.zip)

I notice that the "goto" attempts trigger the creation of a new Arduino Language Server temporary folder, which breaks the URIs I see for `src/foo.h` in the language server logs (e.g., `file:///C:/Users/per/AppData/Local/Temp/arduino-language-server410042786/sketch/src/foo.h`).

per1234, 2022-03-11:
There is additional discussion, which gets into speculation about the cause of the problem at https://github.com/arduino/arduino-ide/issues/323

mink99, 2022-03-17:
I added another simple example :
src :
`
#include "inc.h"
void setup() {
// put your setup code here, to run once:
Serial.begin(9600);
}

void loop() {
// put your main code here, to run repeatedly:

}
included : inc.h
void doit()
{
Serial.println("epic fail")
}
`
result :
Serial is known in the main , but is a clang error in "inc.h"
[image]

per1234, 2022-03-17:
I don't think that is the same problem @mink99.

The Arduino sketch precompiler automatically adds an `#include` directive to the `sketch_mar17a.ino`. That provides the declaration of `Serial`. But that is only done for `.ino` files. There is no sketch preprocessing for `.h` files, so `Serial` is indeed undeclared in that code.

You will get the same even if you run clangd directly on the equivalent C++ program.

If you want to use the Arduino API in the sketch code files that don't have a `.ino` file extension, you should add the `#include` directive to them:

```cpp
#include <Arduino.h>
```