arduino/arduino-ide #724: "Find Next" has no effect

per1234, 2021-12-27:
## Describe the bug

The **Edit > Find Next** menu item is intended to find the next occurrence of the chosen text.

:bug: It currently has no effect.

## To Reproduce

1. Select **File > New** from the Arduino IDE menus.
1. Select **Edit > Find** from the Arduino IDE menus.
   The "**Find**" UI opens with `void` pre-filled.
1. Select **Edit > Find Next** from the Arduino IDE menus.
   :bug: Nothing happens.
1. Select **Edit > Find Next** from the Arduino IDE menus.
   :bug: Nothing happens.
1. Select **Edit > Find Previous** from the Arduino IDE menus.
   :slightly_smiling_face: The occurrence of `void` at line 1 is selected.
1. Select **Edit > Find Previous** from the Arduino IDE menus.
   :slightly_smiling_face: The occurrence of `void` at line 6 is selected.
1. In the "**Find**" UI, click the **↓** icon.
   :slightly_smiling_face: The occurrence of `void` at line 1 is selected.

## Expected behavior

The same behavior from **Edit > Find Next** as from **Edit > Find Previous**, except in the opposite direction.

## Desktop

- OS: Windows 10
- IDE Version: 2.0.0-rc3
Date: 2021-12-22T15:46:56.004Z
- Arduino CLI Version: 0.20.2 [13783819]

ubidefeo, 2022-02-16:
@per1234 
I have tested this one on a recent PR build and it is working form the UI's arrows but the shortcut (CMD+G on Mac) is not honoured

ubidefeo, 2022-02-17:
investigate if it's a bug/regression from Monaco editor, since when hitting the shortcut the popup opens with `? of X` (see screenshot)
/Users/ubidefeo/Library/Application Support/CleanShot/media/media_ky5DPu4DMW/CleanShot 2022-02-17 at 11.55.31@2x.png[image]

per1234, 2022-02-17:
>it is working form the UI's arrows

Same here, as I reported in step (7) in the instructions.

>the shortcut (CMD+G on Mac) is not honoured

Same for me on Windows (<kbd>Ctrl</kbd>+<kbd>G</kbd>). I notice that there is a shortcut collision with the default shortcut for **Go to Line/Column...** (did anyone bother to check this stuff?), but even after I change **Go to Line/Column...** shortcut to something else <kbd>Ctrl</kbd>+<kbd>G</kbd> still doesn't do a "**Find Next**". <kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>G</kbd> works just fine for "**Find Previous**".

> when hitting the shortcut the popup opens with `? of X`

@ubidefeo which shortcut?

@ubidefeo does it also happen with the **File > New** sketch, or only with that "NiclaShopSensor" sketch?

per1234, 2022-03-16:
Closing as fixed by https://github.com/arduino/arduino-ide/commit/ca1c24050d15bf43b3a5a97af6a294bbd56df1b3

Thanks @msujew!

closed 2022-03-16 by per1234