arduino/arduino-ide #726: Upload fails in subsequent windows with Serial Monitor open

per1234, 2021-12-28:
## Describe the bug

The Arduino IDE uses the serial ports of Arduino boards in two different ways:

- Uploading sketches
- Communication via Serial Monitor

The port can not be used for both of these things simultaneously. The IDE handles the condition of Serial Monitor being open when an upload is started automagically like so:

1. Close port in Serial Monitor
1. Complete upload process
1. Reopen port in Serial Monitor

:bug: This works as expected in the first IDE window you open, but it no longer works in subsequent windows.

***NOTE***: although similar, this is different from https://github.com/arduino/arduino-ide/issues/586 because it occurs even when the port is not open in a Serial Monitor/Serial Plotter of another IDE window.

## To Reproduce

1. Connect an Arduino board to your computer.
1. Select the appropriate board and port.
1. Select **Tools > Serial Monitor** from the Arduino IDE menus to open Serial Monitor.
1. Select **Sketch > Upload** from the Arduino IDE menus.
   :slightly_smiling_face: Upload succeeds as expected.
1. Select **Tools > Serial Monitor** from the Arduino IDE menus to close Serial Monitor.
   This is necessary to avoid https://github.com/arduino/arduino-ide/issues/586
1. Select **File > New** from the Arduino IDE menus.
1. Select **Tools > Serial Monitor** from the Arduino IDE menus in the new window.
    :bug: Note that there is an unexpected banner in Serial Monitor:
   > Not connected. Select a board and a port to connect automatically.
1. Select **Sketch > Upload** from the Arduino IDE menus.
   :bug: Upload fails with an error something like "avrdude: ser_open(): can't open device "\\.\COM7": Access is denied."

## Expected behavior

Upload with Serial Monitor open in the same IDE window to always be handled automagically.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc3
Date: 2021-12-22T15:46:56.004Z
CLI Version: 0.20.2 [13783819]

## Additional context

It is not necessary to leave the original window open to reproduce the bug. It occurs even if that window is closed.

---

I bisected the issue to https://github.com/arduino/arduino-ide/commit/767b09d2f10cab78b07754039ebb749d55b45d8d (it does not occur at https://github.com/arduino/arduino-ide/commit/88397931c5b704516e881959f6e7ce106094d14d).

---

Originally reported at https://forum.arduino.cc/t/upload-failed-when-serial-monitor-is-open/940255

closed 2022-06-08 by per1234

ProxyPlayerHD, 2022-08-14:
why is this closed? this is still an issue in rc9.2.
If you don't manually close the "Serial Monitor" tab at the bottom before trying to upload a sketch, it just gives you the:
`Failed uploading: uploading error: exit status 1avrdude: ser_open(): can't open device "\\.\COM#": Access Denied`
error

per1234, 2022-08-14:
Hi @ProxyPlayerHD. Thanks for your report.

The issue is closed because the bug was fixed for me and the majority of the other users who reported being affected by it after the changes made in https://github.com/arduino/arduino-ide/pull/982.

We are tracking one other report of continued failing uploads at https://github.com/arduino/arduino-ide/issues/1306, but have not been able to determine how the developers can reproduce that issue.

Please provide more information:

- Which Arduino board are you using?
- Does the problem occur consistently, or intermittently?
- Does the problem persist after you restart your computer?
- Does the problem also occur when using Arduino IDE 2.0.0-rc9 or older (download links for 2.0.0-rc9 listed [here](https://github.com/arduino/arduino-ide/releases/tag/2.0.0-rc9#:~:text=Assets), or 1.8.19 [here](https://www.arduino.cc/en/software))

ProxyPlayerHD, 2022-08-14:
hmm, i hate it when it does this... aparently the issues seems to have fixed itself out of nowhere. but now a different issue emerged where the Serial Monitor doesn't display anything from my Arduino Mega 2560 but works fine for other board i tested it with (Nucleo 144, and RaspPi Pico).
i just have a looping "Hello World!" program on the Arduino and i can see the sending LED constantly being lit up but nothing appears in the IDE. I can send stuff from the Serial Monitor to the Arduino and see the receiving LED light up shortly when i do it, so clearly it has a working connection.
power cycling/reconnecting the Arduino or restarting the IDE doesn't fix it. (reuploading the sketch while the Serial Monitor is open makes it print a few garbled "Hello World"'s)
i can also use some other terminal software like Teraterm to connect to the Arduino and it works perfectly fine there.

this is probably enough to open a seperate issue, but i'll restart my PC to see if that maybe fixes it

chrisAS12, 2023-03-10:
Still am having this issue, it does not seem to be fixed.

chrisAS12, 2023-03-10:
It's a custom PCB, but it's compatible with Arduino Uno. Using FTDI adapter for uploading.

per1234, 2023-03-10:
>Using FTDI adapter for uploading.

Thanks! That's the exact information I was angling for. We are tracking the bug here:
https://github.com/arduino/arduino-ide/issues/1783