arduino/arduino-ide #729: #include <IRremote.hpp> does not work in 2.0.0-rc3 but works in 1.8.16 (and before)

ArminJo, 2021-12-30:
**Describe the bug**
Compiling a standard example of the IRremote library leads to:
```
Detecting libraries used...
"C:\\Users\\Armin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Users\\Armin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.4\\cores\\arduino" "-IC:\\Users\\Armin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.4\\variants\\standard" "C:\\Users\\Armin\\AppData\\Local\\Temp\\arduino-sketch-ED5555B020A1FEA4472DB8AE5E8CD203\\sketch\\SimpleSender.ino.cpp" -o nul
Alternatives for IRremote.hpp: []
ResolveLibrary(IRremote.hpp)
  -> candidates: []
C:\Users\Armin\AppData\Local\Temp\.arduinoIDE-unsaved20211130-15880-71zsuz.v72w\SimpleSender\SimpleSender.ino:25:10: fatal error: IRremote.hpp: No such file or directory
 #include <IRremote.hpp>
          ^~~~~~~~~~~~~~
compilation terminated.
Compilation error: exit status 1}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install IRremote library
2. Select File > Examples > IRremote > SimpleSender
3. Verify
4. See error

**Expected behavior**
Should work like in the 1.8.x IDE

**Desktop (please complete the following information):**
 - OS: Windows11

per1234, 2022-01-03:
Hi @ArminJo. Thanks for your report.

I was not able to reproduce this issue. Which version of the IRremote library do you have installed (I have the latest release 3.5.2)? Did you install the library via Library Manager, or manually?

ArminJo, 2022-01-03:
I used the download manager, and the IDE was able to find the examples.
But **I changed the `Sketchbook location` of the IDE** and this caused the error, since reinstalling the library placed another copy under the new `Sketchbook location` /libraries and there it could be found.
So it seems that you loose all your installed libraries when changing `Sketchbook location` of the IDE.

per1234, 2022-01-03:
Thanks for the update.

>it seems that you loose all your installed libraries when changing Sketchbook location of the IDE.

That is correct. The path of the globally installed Arduino libraries is relative to the sketchbook location. So when you change the "Sketchbook location" preference, the libraries installed under the previous location are no longer accessible to the IDE. Some advanced users even use multiple sketchbooks in order to control the dependencies (which can include boards platforms under the `hardware` subfolder in addition to libraries).

The IDE currently has a confusing behavior in that the menu contents are not refreshed after the "Sketchbook location" preference is changed, so it appears that the libraries in the old location are still in effect even though they aren't. That is being tracked here: https://github.com/arduino/arduino-ide/issues/570

closed 2022-01-03 by per1234