arduino/arduino-ide #730: Text copied from from Serial Monitor has missing or non-standard line endings

per1234, 2021-12-31:
## Describe the bug

It is sometimes useful to copy the text that was printed by an Arduino board to Serial Monitor. For example, I sometimes copy CSV-style data into a spreadsheet program when I want to do further manipulation and analysis.

:bug: This is currently not possible to do with any multi-line text because the copied content has missing (or perhaps non-standard) line endings.

## To Reproduce

1. Upload a sketch that prints multiple lines of text to serial:
   ```cpp
   void setup() {
     Serial.begin(9600);
   }
   void loop() {
     Serial.println("hello");
     delay(1000);
   }
   ```
1. Open the Serial Monitor view in the Arduino IDE (**Tools > Serial Monitor**).
1. Select multiple lines of the text that was printed in the Serial Monitor view.
1. Paste it into a text editor or spreadsheet program.
   :bug: The line endings are not present:
   ```
   hellohellohello
   ```

## Expected behavior

Text copied from Serial Monitor has standard line endings.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc3
Date: 2021-12-22T15:46:56.004Z
CLI Version: 0.20.2 [13783819]

## Additional context

It seems that there is some form of line break formatting in the text because I do get line breaks when I paste the text to the Arduino Forum (Discourse) composer field. In that case, the problem is opposite because I get double spaced text. But when pasting to VS Code, Notepad++, LibreOffice Calc, GitHub comment field, etc. I get no line breaks.

---

Related:

- https://github.com/arduino/arduino-ide/issues/105

---

Originally reported at:

- https://forum.arduino.cc/t/serial-monitor/698844
- https://forum.arduino.cc/t/serial-monitor-text-copying-not-quite-right/938730
- https://forum.arduino.cc/t/each-day-a-new-problem/937997/3
- https://github.com/arduino/arduino-ide/issues/427#issuecomment-885018751

KurtE, 2022-02-03:
I just started playing with the IDE 2 and ran into this as well.

This happens with Teensy Micromod connected as hardware COM Port COM115.  Will try it again once the plugable Serial monitor code is working and using the Teensy specific plugin.

per1234, 2022-06-22:
Fixed by https://github.com/arduino/arduino-ide/pull/943

closed 2022-06-22 by per1234