arduino/arduino-ide #735: No icon in the latest version

dominpiano, 2022-01-04:
There is no app icon in the latest version of Arduino 2.0.0, and there is no possibility to adjust the height of the output.

per1234, 2022-01-04:
Thanks for your report @dominpiano.

We are going to need more details in order to investigate this.

Which operating system are you using?

>there is no possibility to adjust the height of the output.

What do you mean by "output"?

If you mean the bottom panel where the Output view is, this is how it works for me:

[image]

Are you experiencing something different from that? If so, please provide a detailed description. Include screencast or screenshot if appropriate.

dominpiano, 2022-01-05:
Okay, this was some temporary bug, maybe from first open of the program, now I have this possibility. Probably I have no icon on this app because the same issue of some problems during installation. 
Thanks a lot

closed 2022-01-05 by dominpiano